dividuum/info-beamer #22: CRITICAL error: Cannot open window

nichdu, 2015-01-05:
When I try to start info-beamer on my VirtualBox Debian 7.7-amd64 virtual machine info-beamer exits with just the above error message, see the screen shot: 
[image]
I installed all dependencies and enabled OpenGL support for my virtual machine. I'm not really familiar with C, so I can't check the errors myself.

dividuum, 2015-02-27:
Sorry for the delay. I guess you're running this inside some X terminal windows? Is the DISPLAY environment variable set? Can you run glxinfo or glxgears? Do they produce any output?

nichdu, 2015-03-02:
Yes, I'm running this in an uxterm.
The output of $DISPLAY, glxinfo and glxgears is in https://gist.github.com/nichdu/886013a6bfe385037086
Obviously glxgears produces an output.

dividuum, 2015-03-02:
Is seems that glfw had some issues with Chromium (the GL implementation in VirtualBox): https://github.com/glfw/glfw/blob/master/src/glx_context.c#L55

Which version of glfw are you using? Can you install glfw3 and use the glfw3 branch of info-beamer? Maybe that solves the problem.

dividuum, 2015-03-02:
Maybe, but that's pure guessing, you can try to use fullscreen mode?

    INFOBEAMER_FULLSCREEN=1 ./info-beamer samples/shader

snowborn, 2016-01-11:
Hi there!

Seems I am running into the same problem with Ubuntu 15.10 in Virtualbox.

When starting the "hello" example, it says: 
libGL error: core dri or dri2 extension not found
libGL error: failed to load driver: vboxvideo
[main.c] initial size is 1024x768
CRITICAL ERROR: need opengl 3.0 support

glxinfo returns this: (it uses the Chromium Engine as well)
[test.txt](https://github.com/dividuum/info-beamer/files/86381/test.txt)

Running in fullscreen did not help. Where do I find the branch for glfw3 for infobeamer? Or any other suggestion to make it run now. :)

dividuum, 2016-01-12:
The glfw3 branch is merged to master, so you're already using it. You could try to remove the line in main.c that raises the crititcal error and see how far you get. But I guess you might run in the problems described in #21 then. Worth a shot anyway, I guess.

snowborn, 2016-01-12:
...
libGL error: failed to load driver: vboxvideo
[main.c] initial size is 1024x768
[hello] update +README.silkscreen.txt
[hello] update +silkscreen.ttf
[hello] update +README.txt
[hello] kernel: booting node
[main.c] initialization completed
Speicherzugriffsfehler (Speicherabzug geschrieben)

Does not really seem to help :(

Any other ideas?

dividuum, 2016-01-13:
Sorry. Don't have any. Why are you running info-beamer in Virtualbox? Are you using OS X? I imagine porting info-beamer to OSX isn't too difficult. The one thing that needs replacing is the inotify part, but other than that I guess all dependencies should be available on OS X. I haven't looked into that, since I don't own a Mac, but maybe someone else will?

snowborn, 2016-01-13:
Well, there is a Windows PC running at the place where I want to use the software so I tried running it on that.

snowborn, 2016-01-13:
Ok, found the solution. There is the possibility to use 3D acceleration for the VM. That has to be turned off for it in order to run. :)
Thanks for the speedy support. =D

dividuum, 2016-01-14:
I'm really confused now. After turning 3D acceleration off, what happens? Is it using a software renderer? Is that fast enough?

PowerPan, 2017-01-17:
I had the same on my VM runnig Ubnutu 16.10
Now The UI is slower, after disabling 3D acceleration, but info-beamer is working now

closed 2017-02-07 by dividuum