dividuum/info-beamer #23: Does not build on sabayon/gentoo

jamasi, 2015-01-22:
I tried to compile info-beamer on sabayon (result is expected to be the same on gentoo). I installed glfw-3.0.4 and devil-1.7.8 from portage and then cloned the repository.

First thing I noticed:
    #include <GL/glfw.h>
should rather be 
    #include <GLFW/glfw3.h>

but even then I get these errors:
https://gist.github.com/jamasi/b87bf40e69c9610e3a60

    main.c:1159:22: Fehler: expected »=«, »,«, »;«, »asm« or »__attribute__« before »reshape«
     static void GLFWCALL reshape(int width, int height) {
                          ^

dividuum, 2015-01-23:
Can you try the glfw3 branch?

jamasi, 2015-01-25:
Works, after applying this little patch:

    diff --git a/Makefile b/Makefile
    index 37dca9f..33a2436 100644
    --- a/Makefile
    +++ b/Makefile
    @@ -31,7 +31,7 @@ endif
     
     CFLAGS  += -DVERSION='"$(VERSION)"'
     CFLAGS  += $(LUA_CFLAGS) -I/usr/include/freetype2/ -I/usr/include/ffmpeg -std=c99 -Wall
    -LDFLAGS += $(LUA_LDFLAGS) -levent -lglfw3 -lGL -lGLU -lGLEW -lftgl -lIL -lILU -lavformat -lavcodec -lavutil -lswscale -lz -lm -lXi -lX11 -lXxf86vm -lXrandr -lpthread
    +LDFLAGS += $(LUA_LDFLAGS) -levent -lglfw -lGL -lGLU -lGLEW -lftgl -lIL -lILU -lavformat -lavcodec -lavutil -lswscale -lz -lm -lXi -lX11 -lXxf86vm -lXrandr -lpthread
     
     prefix                 ?= /usr/local
     exec_prefix ?= $(prefix)

dividuum, 2015-01-29:
Wait. GLFW version 3 is linked as glfw on gentoo? Good to know. Thanks for your report. Not sure how I'll handle this without changing the simple make based build system.

closed 2015-01-29 by dividuum