dividuum/info-beamer #24: Add option to rotate the whole screen

jamasi, 2015-02-27:
Maybe I've missed the relevant part of the documentation, but I wonder if there is an environment variable to tell info-beamer that the display screen is oriented in portrait rather than in landscape mode. Also the option to mirror the image in either direction might be useful for back projection displays.

dividuum, 2015-02-27:
There is no environment variable for this right now. But you can setup the surface yourself:

```lua
gl.setup(NATIVE_WIDTH, NATIVE_HEIGHT)

function rotate(degree)
    if degree == 0 then
        return function() end
    elseif degree == 90 then
        WIDTH, HEIGHT = HEIGHT, WIDTH
        return function()
            gl.translate(HEIGHT, 0)
            gl.rotate(90, 0, 0, 1)
        end
    elseif degree == 180 then
        return function()
            gl.translate(WIDTH, HEIGHT)
            gl.rotate(180, 0, 0, 1)
        end
    elseif degree == 270 then
        WIDTH, HEIGHT = HEIGHT, WIDTH
        return function()
            gl.translate(0, WIDTH)
            gl.rotate(-90, 0, 0, 1)
        end
    else
        error("unsupported rotation")
    end
end

function mirror(fn)
    return function()
        fn()
        gl.translate(WIDTH, 0)
        gl.scale(-1, 1)
    end
end

local screen_setup = mirror(rotate(270))
-- local screen_setup = rotate(90)
-- local screen_setup = rotate(0)

local font = resource.load_font("silkscreen.ttf")

function node.render()
    screen_setup()
    font:write(0, 0, "Hello World", 100, 1,1,1,1)
end
```

Does this work for you?

jamasi, 2015-03-05:
Seems to be working fine here. Thank you. (Would be good to add this to the documentation.)

dividuum, 2015-03-05:
I'll probably include it in a future info-beamer release, so it can be used without adding anything to your code. Thanks for testing.

closed 2015-03-05 by dividuum