dividuum/info-beamer #25: glGetError 0x505 on info-beamer PI 0.8.3-beta.56735f

jamasi, 2015-03-05:
I'm in the process of evaluating info-beamer-pi for our association. So (based on example code) I put together this visualisation:

```lua
gl.setup(NATIVE_WIDTH, NATIVE_HEIGHT)

function rotate(degree)
    if degree == 0 then
        return function() end
    elseif degree == 90 then
        WIDTH, HEIGHT = HEIGHT, WIDTH
        return function()
            gl.translate(HEIGHT, 0)
            gl.rotate(90, 0, 0, 1)
        end
    elseif degree == 180 then
        return function()
            gl.translate(WIDTH, HEIGHT)
            gl.rotate(180, 0, 0, 1)
        end
    elseif degree == 270 then
        WIDTH, HEIGHT = HEIGHT, WIDTH
        return function()
            gl.translate(0, WIDTH)
            gl.rotate(-90, 0, 0, 1)
        end
    else
        error("unsupported rotation")
    end
end

function mirror(fn)
    return function()
        fn()
        gl.translate(WIDTH, 0)
        gl.scale(-1, 1)
    end
end

-- local screen_setup = mirror(rotate(270))
local screen_setup = rotate(270)
-- local screen_setup = rotate(0)

-- local font = resource.load_font("silkscreen.ttf")

local distort_shader = resource.create_shader([[
    uniform sampler2D Texture;
    varying vec2 TexCoord;
    uniform vec4 Color;
    uniform float effect;
    void main() {
        vec2 uv = TexCoord.st;
        vec4 col;
        col.r = texture2D(Texture,vec2(uv.x+effect+sin(uv.y*20.0*effect)*0.02,uv.y)).r;
        col.g = texture2D(Texture,vec2(uv.x+effect+sin(uv.y*30.0*effect)*0.02,uv.y)).g;
        col.b = texture2D(Texture,vec2(uv.x-effect+sin(uv.y*40.0*effect)*0.02,uv.y)).b;
        col.a = texture2D(Texture,vec2(uv.x,uv.y)).a;
        gl_FragColor = Color * col;
    }
]])

local images
local current_image_idx

function alphanumsort(o)
    local function padnum(d) return ("%03d%s"):format(#d, d) end
    table.sort(o, function(a,b)
        return tostring(a):gsub("%d+",padnum) < tostring(b):gsub("%d+",padnum) 
    end)
    return o
end

function init_images()
    images = {}
    for name, _ in pairs(CONTENTS) do
        if name:match(".*jpg") or name:match(".*png") then
            images[#images+1] = name
        end
    end
    current_image_idx = 0
    images = alphanumsort(images)
    -- pp(images)
end

node.event("content_update", init_images)
node.event("content_remove", init_images)

init_images()

function PicRotation(x1, y1, x2, y2, time)
    local image
    function next_image()
        current_image_idx = current_image_idx + 1
        if current_image_idx > #images then
            current_image_idx = 1
        end
        image = resource.load_image(images[current_image_idx])
        switch = sys.now() + time
    end

    next_image()

    local switch_time = 0.2

    function draw()
        local remaining = switch - sys.now()
        local abs = 10
        if remaining < switch_time then 
            abs = switch_time - remaining
        end
        if remaining > time - switch_time then
            abs = remaining - time + switch_time
        end
        if abs < switch_time then
            abs = abs / switch_time
            distort_shader:use{
                effect = abs * 10 * math.sin(sys.now()) * 3;
            }
        end
            util.draw_correct(image, x1, y1, x2, y2)
        distort_shader:deactivate()
        if sys.now() > switch then
            next_image()
        end
    end

    return {
        draw = draw;
    }
end

-- function PicRotation(x1, y1, x2, y2, time)
local pic = PicRotation(0, 0, WIDTH, HEIGHT, 8)

function node.render()
    screen_setup()
    pic:draw()
end
```

In the directory there are 6 .png images each about 1.1 MB in size and 1050x1680 / 1050x1650 in resolution. On my desktop with ```Info Beamer 1.0pre4.fca06c+Lua 5.1``` the visualisation works as expected, but with the Pi 1 B (with gpu_mem=192) I encounter this ```glGetError 0x505``` error:

```
pi@raspberrypi ~/pLAN2 $ INFOBEAMER_FULLSCREEN=1 ../info-beamer-pi/info-beamer  .
Info Beamer PI 0.8.3-beta.56735f (https://info-beamer.com)
Copyright (c) 2015, Florian Wesch <fw@dividuum.de>

info-beamer for the raspberry pi is free for private use. If you
would like to make use of this software in a commercial way, you
must buy a license which will support the continued development.
See https://info-beamer.com/pi for more information.  Thank you!

[main.c] root node is pLAN2
[main.c] loading nodes
[pLAN2] update +02_Gamingcafe.png (search)
[pLAN2] update +04_Socialmediashit.png (search)
[pLAN2] update +03_PreislisteLAN.png (search)
[pLAN2] update +06_NextLAN.png (search)
[pLAN2] update +01_PreislisteLAN.png (search)
[pLAN2] update +05_PreislisteLAN.png (search)
[pLAN2] update +node.lua (search)
[image.c] loading pLAN2/01_PreislisteLAN.png
[main.c] initialization completed
[image.c] loading pLAN2/01_PreislisteLAN.png

    uptime 10s, cpu 1s+0s, rss 20840kb, temp=49.2'C
   mem   fps  width height    cpu flag  name (alias)
-----------------------------------------------------
 115kb  59.0   1680   1050   2.6%  R-s  '- pLAN2 (-)
----------------------------------------------------

[image.c] loading pLAN2/02_Gamingcafe.png

    uptime 20s, cpu 2s+0s, rss 20980kb, temp=50.8'C
   mem   fps  width height    cpu flag  name (alias)
-----------------------------------------------------
 119kb  59.5   1680   1050   1.6%  R-s  '- pLAN2 (-)
----------------------------------------------------

[image.c] loading pLAN2/03_PreislisteLAN.png

    uptime 30s, cpu 2s+0s, rss 20980kb, temp=51.4'C
   mem   fps  width height    cpu flag  name (alias)
-----------------------------------------------------
 122kb  59.5   1680   1050   1.4%  R-s  '- pLAN2 (-)
----------------------------------------------------

[image.c] loading pLAN2/04_Socialmediashit.png

    uptime 40s, cpu 3s+1s, rss 21064kb, temp=51.9'C
   mem   fps  width height    cpu flag  name (alias)
-----------------------------------------------------
 124kb  59.5   1680   1050   1.5%  R-s  '- pLAN2 (-)
----------------------------------------------------

[image.c] loading pLAN2/05_PreislisteLAN.png
[image.c] loading pLAN2/06_NextLAN.png

    uptime 50s, cpu 4s+1s, rss 21064kb, temp=51.4'C
   mem   fps  width height    cpu flag  name (alias)
-----------------------------------------------------
 129kb  59.0   1680   1050   1.8%  R-s  '- pLAN2 (-)
----------------------------------------------------

[image.c] loading pLAN2/01_PreislisteLAN.png

    uptime 60s, cpu 5s+1s, rss 21064kb, temp=53.0'C
   mem   fps  width height    cpu flag  name (alias)
-----------------------------------------------------
 131kb  59.4   1680   1050   1.4%  R-s  '- pLAN2 (-)
----------------------------------------------------

[image.c] loading pLAN2/02_Gamingcafe.png

[...]

[image.c] loading pLAN2/06_NextLAN.png

    uptime 150s, cpu 13s+3s, rss 21112kb, temp=55.1'C
   mem   fps  width height    cpu flag  name (alias)
-----------------------------------------------------
 156kb  59.4   1680   1050   1.4%  R-s  '- pLAN2 (-)
----------------------------------------------------

[image.c] loading pLAN2/01_PreislisteLAN.png

    uptime 160s, cpu 14s+3s, rss 21112kb, temp=55.1'C
   mem   fps  width height    cpu flag  name (alias)
-----------------------------------------------------
 162kb  59.5   1680   1050   1.5%  R-s  '- pLAN2 (-)
----------------------------------------------------

[image.c] loading pLAN2/02_Gamingcafe.png
[image.c] loading pLAN2/03_PreislisteLAN.png
glGetError 0x505
[pLAN2] runtime error: pLAN2/node.lua:121: image loaders failed to parse this image format
stack traceback:
        [C]: in function 'draw_correct'
        pLAN2/node.lua:121: in function 'draw'
        pLAN2/node.lua:139: in function <pLAN2/node.lua:137>
        bundled userlib.lua:616: in function <bundled userlib.lua:613>

    uptime 170s, cpu 15s+4s, rss 21128kb, temp=54.6'C
   mem   fps  width height    cpu flag  name (alias)
-----------------------------------------------------
 168kb  59.2   1680   1050   1.9%  REs  '- pLAN2 (-)
----------------------------------------------------

    uptime 180s, cpu 16s+4s, rss 21128kb, temp=53.0'C
   mem   fps  width height    cpu flag  name (alias)
-----------------------------------------------------
 169kb  60.0   1680   1050   3.0%  REs  '- pLAN2 (-)
----------------------------------------------------

[...]

[main.c] terminating...
[main.c] That's all. Have a nice day
```

Might there be some problem with freeing (or reusing) oGL texture memory?

dividuum, 2015-03-05:
> Might there be some problem with freeing (or reusing) oGL texture memory?

That's probably the reason. I guess I didn't update that script yet. Sorry about that. 
Garbage collection changed a bit in recent versions of info-beamer. See https://info-beamer.com/blog/understanding-garbage-collection to learn more about that.

The quick fix is to put a `node.set_flag("slow_gc", false)` at the top of the lua file. This will force garbage collection after each frame which should help.

jamasi, 2015-03-05:
For your information the vanilla "ordered images" script from info-beamer-nodes exhibits the same problem.

But setting the GC to frame mode fixes the issue for now. Thanks for your quick response.

closed 2015-03-05 by jamasi

dividuum, 2015-03-05:
Thanks for confirming that it works. I'll update the ordered image example.

dividuum, 2015-03-05:
I just pushed a new version to dividuum/info-beamer-nodes@81fc1e8. Have a look in photos-fade. It completely replaces ordered-images and on top of not leaking image resources it also supports cross-fading between images.