dividuum/info-beamer #27: CRITICAL ERROR: cannot create surface (4k@rpi)

yetzt, 2015-06-10:
i have a raspberry pi running with 3840×2160,16@28.

```
pi@weywot ~ $ fbset -s
mode "3840x2160"
    geometry 3840 2160 3840 2160 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/16
endmode
```
info-beamer does not work:

```
pi@weywot ~/info-beamer-pi $ INFOBEAMER_LOG_LEVEL=9 ./info-beamer ./samples/hello/
Info Beamer PI 0.9.0-beta-pre7.b22773 (https://info-beamer.com)
Copyright (c) 2015, Florian Wesch <fw@dividuum.de>

info-beamer for the raspberry pi is free for private use. If you
would like to make use of this software in a commercial way, you
must buy a license which will support the continued development.
See https://info-beamer.com/pi for more information.  Thank you!

[main.c] chdir /home/pi/info-beamer-pi/samples
[main.c] root node is hello
[main.c] tcp/udp port is 4444
[main.c] bcm_host_init
[main.c] luajit version: LuaJIT 2.1.0-alpha~1e67a4
[main.c] linux version: 3.18.11-v7+ #777 SMP PREEMPT Sat Apr 11 17:30:37 BST 2015 armv7l
[main.c] firmware version: Apr 11 2015 17:39:13  Copyright (c) 2012 Broadcom version 311b05fb4e33655a083a6f65b645e6a14e322803 (clean) (release)
[main.c] config: arm_control=0xa5800010 arm_freq=900 avoid_fix_ts=1 config_hdmi_boost=2 disable_commandline_tags=2 disable_l2cache=1 disable_overscan=1 emmc_pll_core=1 force_eeprom_read=1 force_pwm_open=1 framebuffer_height=2160 framebuffer_ignore_alpha=1 framebuffer_swap=1 framebuffer_width=3840 hdmi_force_cec_address=65535 hdmi_group=2 hdmi_ignore_edid=0xa5000080 hdmi_mode=87 hdmi_pixel_freq_limit=0x17d78400 ignore_lcd=1 max_framebuffer_height=2160 max_framebuffer_width=3840 over_voltage_avs=0x1b774 pause_burst_frames=1 program_serial_random=1 sdram_freq=450 temp_limit=85
[main.c] memory: arm=816M gpu=192M
[main.c] file descriptor limit: 1024
[surface.c] config  1: 8,8,8,8 depth=24,stencil= 8, samples=0, sample_buffers=0
[surface.c] config  2: 8,8,8,0 depth=24,stencil= 8, samples=0, sample_buffers=0
[surface.c] config  3: 8,8,8,8 depth=24,stencil= 0, samples=0, sample_buffers=0
[surface.c] config  4: 8,8,8,0 depth=24,stencil= 0, samples=0, sample_buffers=0
[surface.c] config  5: 8,8,8,8 depth= 0,stencil= 8, samples=0, sample_buffers=0
[surface.c] config  6: 8,8,8,0 depth= 0,stencil= 8, samples=0, sample_buffers=0
[surface.c] config  7: 8,8,8,8 depth= 0,stencil= 0, samples=0, sample_buffers=0
[surface.c] config  8: 8,8,8,0 depth= 0,stencil= 0, samples=0, sample_buffers=0
[surface.c] config  9: 8,8,8,8 depth=24,stencil= 8, samples=4, sample_buffers=1
[surface.c] config 10: 8,8,8,0 depth=24,stencil= 8, samples=4, sample_buffers=1
[surface.c] config 11: 8,8,8,8 depth=24,stencil= 0, samples=4, sample_buffers=1
[surface.c] config 12: 8,8,8,0 depth=24,stencil= 0, samples=4, sample_buffers=1
[surface.c] config 13: 8,8,8,8 depth= 0,stencil= 8, samples=4, sample_buffers=1
[surface.c] config 14: 8,8,8,0 depth= 0,stencil= 8, samples=4, sample_buffers=1
[surface.c] config 15: 8,8,8,8 depth= 0,stencil= 0, samples=4, sample_buffers=1
[surface.c] config 16: 8,8,8,0 depth= 0,stencil= 0, samples=4, sample_buffers=1
[surface.c] config 17: 5,6,5,0 depth=24,stencil= 8, samples=0, sample_buffers=0
[surface.c] config 18: 5,6,5,0 depth=24,stencil= 0, samples=0, sample_buffers=0
[surface.c] config 19: 5,6,5,0 depth= 0,stencil= 8, samples=0, sample_buffers=0
[surface.c] config 20: 5,6,5,0 depth= 0,stencil= 0, samples=0, sample_buffers=0
[surface.c] config 21: 5,6,5,0 depth=24,stencil= 8, samples=4, sample_buffers=1
[surface.c] config 22: 5,6,5,0 depth=24,stencil= 0, samples=4, sample_buffers=1
[surface.c] config 23: 5,6,5,0 depth= 0,stencil= 8, samples=4, sample_buffers=1
[surface.c] config 24: 5,6,5,0 depth= 0,stencil= 0, samples=4, sample_buffers=1
[surface.c] config 25: 8,8,8,8 depth= 0,stencil= 0, samples=0, sample_buffers=0
[surface.c] config 26: 8,8,8,0 depth= 0,stencil= 0, samples=0, sample_buffers=0
[surface.c] config 27: 5,6,5,0 depth= 0,stencil= 0, samples=0, sample_buffers=0
[surface.c] config 28: 5,6,5,0 depth=16,stencil= 0, samples=0, sample_buffers=0
[surface.c] egl extensions: EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_vg_parent_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_lock_surface
[surface.c] display configs matching: 1. using config 7
[surface.c] screen size: 3840x2160, gl surface 3840x2160+0,0 (layer 0)
[surface.c] creating black layer on -10
CRITICAL ERROR: cannot create surface
```
fyi. :)

dividuum, 2015-06-10:
Oh. I didn't even know that the Pi can use resolutions like that. It looks like the OpenGL surface cannot be bigger than (probably) 2048x2048, the maximum texture size. info-beamer uses the screen size as initial gl surface size, so creating the surface fails. Can you have a look at https://usercontent.infobeamer.com/info-beamer-pi-0.9.0-beta.152279.tar.gz

Now the initial surface is small enough, so initialization should work. It is resized later once gl.setup is called.

yetzt, 2015-06-11:
works fine with the new release. :)

dividuum, 2015-06-11:
Thanks for testing.

closed 2015-06-11 by dividuum

yetzt, 2015-06-11:
now you can add "4k compatibility" to your release notes. thanks for development. :)