dividuum/info-beamer #28: Blank output window

vliaskov, 2015-06-18:
Compiled from git master 5ba83177. All samples/ produce a blank window for me. Machine is x86_64 with Ubuntu 14.04, lua 5.1 and graphics card ATI Radeon HD 7850. glxgears works fine.

I don't see any error output, is there a verbose/debugging mode to provide some hints?

/opt/devel/info-beamer$ ./info-beamer samples/hello/
Info Beamer 1.0pre4.5ba831+Lua 5.1 (http://info-beamer.org/)
Copyright (c) 2015 Florian Wesch <fw@dividuum.de>

[main.c] chdir /opt/devel/info-beamer/samples
[main.c] root node is hello
[main.c] tcp/udp port is 4444
[main.c] initial size is 1024x768
[main.c] resized to 1024x768
[hello] update +silkscreen.ttf
[hello] update +README.txt
[hello] update +README.silkscreen.txt
[hello] kernel: booting node
[main.c] initialization completed
[image]

[framebuffer.c] new framebuffer (1024x768): 1
[framebuffer.c] new framebuffer (1024x768): 2

similar output from all other examples in samples/

dividuum, 2015-06-18:
Odd. Can you remove this line: https://github.com/dividuum/info-beamer/blob/master/main.c#L1447 and change the glClearColor (2 lines above) to glClearColor(1,0,0,1). You should see a completely red screen. Does that work?

vliaskov, 2015-06-18:
yes that works, Does that mean the problem is somewhere in the lua nodes processing?

dividuum, 2015-06-18:
Maybe something is wrong with how framebuffers are used. The top-level node renders into a texture which is then drawn onto the available GL surface. You might be able to track the error by adding

```
assert(glGetError() == GL_NO_ERROR);
```

all over the place.

vliaskov, 2015-06-19:
This bug looks the same as https://bugs.freedesktop.org/show_bug.cgi?id=81689
Have you tested on ATI GPUs?

I have used apitrace to collect opengl traces of infobeamer samples/hello
When replaying the trace, everything works fine, but there are errors and warnings in several GL calls. The initial info-beamer run only displays blank window, contrary to the trace replay.

Example errors from apitrace replay:
4105: message: major api error 1: GL_INVALID_ENUM in glTexParameter(param=GL_LINEAR_MIPMAP_LINEAR)
4105 @0 glTexParameteri(target = GL_TEXTURE_2D, pname = GL_TEXTURE_MAG_FILTER, param = GL_LINEAR_MIPMAP_LINEAR)
4105: warning: glGetError(glTexParameteri) = GL_INVALID_ENUM
4123: message: major api error 1: GL_INVALID_VALUE in glGetUniformLocation
4123 @0 glGetUniformLocation(program = 0, name = "Color") = -1
4123: warning: glGetError(glGetUniformLocation) = GL_INVALID_VALUE
4453: message: major api error 1: GL_INVALID_OPERATION in glReadBuffer(buffer=0x405)
4453: message: major api error 1: GL_INVALID_OPERATION in glDrawBuffer(buffer=0x405)
4453 @0 glPopAttrib()
4453: warning: glGetError(glPopAttrib) = GL_INVALID_OPERATION

I have commented out the offending GL calls for testing and the trace replay shows no errors now. But the initial info-beamer still only displays blank output. 
diff --git a/framebuffer.c b/framebuffer.c
index 425158e..335883c 100644
--- a/framebuffer.c
+++ b/framebuffer.c
@@ -51,8 +51,8 @@ void make_framebuffer(int width, int height, GLuint *tex, GLuint *fbo) {

     glGenTextures(1, tex);
     glBindTexture(GL_TEXTURE_2D, *tex);
-    glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_MAG_FILTER, GL_LINEAR_MIPMAP_LINEAR);
-    glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_MIN_FILTER, GL_LINEAR_MIPMAP_LINEAR);
+    //glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_MAG_FILTER, GL_LINEAR_MIPMAP_LINEAR);
+    //glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_MIN_FILTER, GL_LINEAR_MIPMAP_LINEAR);
     glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_WRAP_S, GL_CLAMP);
     glTexParameteri(GL_TEXTURE_2D, GL_TEXTURE_WRAP_T, GL_CLAMP);
     glTexImage2D(GL_TEXTURE_2D, 0, GL_RGBA8, width, height, 0, GL_RGBA, GL_INT, NULL);
diff --git a/main.c b/main.c
index c9e2b27..9d4164d 100644
--- a/main.c
+++ b/main.c
@@ -705,7 +705,7 @@ static int node_render_to_image(lua_State *L, node_t *node) {
     glGetDoublev(GL_PROJECTION_MATRIX, prev_projection);
     glGetDoublev(GL_MODELVIEW_MATRIX, prev_modelview);

-    glPushAttrib(GL_ALL_ATTRIB_BITS);
+    //glPushAttrib(GL_ALL_ATTRIB_BITS);

     int width = 1, height = 1;
     if (node_setup_completed(node))
@@ -758,7 +758,7 @@ static int node_render_to_image(lua_State *L, node_t *node) {
     glGenerateMipmap(GL_TEXTURE_2D);

     // restore previous state
-    glPopAttrib();
+    //glPopAttrib();

     glMatrixMode(GL_PROJECTION);
     glLoadMatrixd(prev_projection);
diff --git a/shader.c b/shader.c
index 053c29a..2cf9931 100644
--- a/shader.c
+++ b/shader.c
@@ -200,7 +200,7 @@ void shader_set_gl_color(GLfloat r, GLfloat g, GLfloat b, GLfloat a) {

     GLint prog, color;
     glGetIntegerv(GL_CURRENT_PROGRAM, &prog);
-    color = glGetUniformLocation(prog, "Color");
+    /*color = glGetUniformLocation(prog, "Color");
     if (color != -1)
-        glUniform4f(color, r, g, b, a);
+        glUniform4f(color, r, g, b, a);*/
 }

dividuum, 2015-06-22:
Hm. I'm using an Nvidia GPU right now and I haven't really tested anything else. I honestly don't know what to do without access to an ATI GPU. I might try get access to a box that has one, but I can't make any promises :-(

vliaskov, 2015-06-23:
This is an issue only with the open source "radeon" driver. Proprietary ATI driver "fglrx" works.

dividuum, 2015-06-23:
Hmpf. Ok. I guess the open source driver is either better at checking errors or API misuse or just buggy. Maybe someone else has more ideas?

sheepherder, 2015-07-08:
I can confirm the blank screen behavior with the radeon driver. fglrx works.
On another machine with Nvidia GPU, I had the same blank screen issue with the open source driver "nouveau". Nvidia proprietary driver works.
On a third machine with Intel GPU, it works with the open source driver. There is no alternate binary intel graphics driver to test.

On all the machines though, there is no text antialising. The raspberry pi version has antialiased text. Does antialiasing not work in the open source version or did I forget to install something? System is debian 8.1 jessie.

dividuum, 2015-07-08:
The Raspberry version uses a completely new way to render fonts, since there was not library available that works with OpenGL ES. This version uses [ftgl](http://ftgl.sourceforge.net/). I might have a look on what has to be changed to make fonts anti aliased.

vliaskov, 2015-07-14:
Branch glfw3 works for radeon. It seems that the  glfwGetFramebufferSize(window, &win_w, &win_h); call (main.c:1561) is needed to correctly initialize framebuffer size with radeon.  Commenting that line out results in blank output also.

Any reason why glfw3 branch is not merged into master? Is glfw3 supported for other platforms e.g. rpi?

dividuum, 2015-07-15:
If you comment out the glfwGetFramebufferSize call I guess there is still output, but it's 0x0 pixel in size :-)

Thanks for letting me know that the glfw3 branch works. The branch isn't merged yet, since I'm still on Ubuntu 14.04 and that's not available as a package there. What's the current state on other distributions? Is glfw3 generally available? Is so, I should probably merge and update as well.

I didn't use glfw3 on the PI yet. [info-beamer pi](https://info-beamer.com/pi) creates a GL context without the help of any libraries. That way I have more control over initialization. Note that the code in this repository won't run on the PI even if glfw3 supports it. Porting info-beamer to the PI was a huge effort, since the PI uses OpenGLES2 while this version uses OpenGL3.

sheepherder, 2015-07-16:
Can confirm glfw3 branch works on radeon.
glfw3 is available in Debian Stable, so it's probably everywhere else too. I had to change the linker flags in the makefile from -lglfw3 to

> LDFLAGS += $(LUA_LDFLAGS) -levent -lglfw [....]

(the file is named libglfw.so.3.0 on debian)

vliaskov, 2015-07-16:
On 14.04 glfw3 is indeed missing from default repos. I had to install it from this ppa:
https://launchpad.net/~keithw/+archive/ubuntu/glfw3

I had to change Makefile also on 14.04, same way that was mentioned in previous post: -lglfw instead of -lgflw3

dividuum, 2015-07-17:
I just merged the glfw3 branch. Thanks @vliaskov for the launchpad hint. Works for me now. Also @sheepherder, I tried improving the font rendering a bit. It's not easily possible to get a perfect quality since ftgl is a bit limited but I think it's better than before. Let me know what you think.

sheepherder, 2015-08-19:
Thanks for the improved font rendering. It's much better, but only if I stay around the prepared font size. Using small fontsize effectively removes antialiasing, using a large font gives ugly large upscaled antialiased pixels around the characters (e.g. in the large temperature display in weather-handdrawn). 

I tried moving ftglSetFontFaceSize() to font_write(). Fonts look good, but frame rate dropped through the floor. So this seems to be an expensive operation. Maybe it's possible to prepare a new FTGLfont for each requested size? On PC Hardware, graphics memory should not be a problem, and reusing the same font in many different sizes does not really happen too often anyway (I think).