dividuum/info-beamer #30: Mirrored image

half-pint, 2015-08-12:
I've installed info-beamer on an Arch Linux, using glfw-3.1.1-1. The image displayed is mirrored
[image]
[image]

dividuum, 2015-08-13:
Sorry for the delay. Can you post the complete output of glxinfo?

half-pint, 2015-08-13:
Thank you very much for your reply. http://paste.debian.net/292021/

dividuum, 2015-08-13:
Thanks for the paste. The gallium driver again. I had another user with a similar problem, so there is a pattern somewhere. I'm not at home right now, so I can't do proper debugging, but I'll look into that mid of next week.

half-pint, 2015-08-13:
That's cool, thanks very much for looking into that

ignisf, 2015-08-15:
Experiencing the same oddity, `glxinfo` output -> https://gist.github.com/ignisf/55cc94fd2dcc072afa86

closed 2015-08-15 by dividuum

ignisf, 2015-08-15:
It's fixed, thanks!

dividuum, 2015-08-15:
Thanks for the feedback. That was an embarrassing bug :-\

half-pint, 2015-08-15:
Thanks again! :)

commit ac815b1f (2015-08-15): fixed missing parameter to image_create. will most likely fix #30