dividuum/info-beamer #32: OSC/UDP commands on PI

deadbutalive, 2015-10-15:
I'm having trouble receiving OSC on raspberry pi 2 v0.9.4.  The short version is I'm trying to send a next command via OSC for the video playlist.  Any guidance is appreciated

dividuum, 2015-10-15:
Make sure that you set the environment variable INFOBEAMER_ADDR=0.0.0.0 if you're trying to reach info-beamer from a remote maschine. Otherwise both udp and tcp only allow connections from localhost. Also make sure that packets reach info-beamer. Set INFOBEAMER_LOG_LEVEL=4 and look for "udp packet" lines. If both of these hints don't help, please elaborate what you're doing.

deadbutalive, 2015-10-16:
victory! thanks

dividuum, 2015-10-16:
I'll update the documentation to make it clear that you have to do that. Thanks for your report.

closed 2015-10-16 by dividuum