dividuum/info-beamer #34: Less verbose logging

tuupola, 2016-03-16:
I like the output of INFOBEAMER_LOG_LEVEL=1. Especially the info about each video file played. It is however a bit too verbose when it is sent to remote syslog. 

Is it somehow possible to configure how often the log info is printed out. I believe the default is 10 seconds. I am looking for something like INFOBEAMER_LOG_DELAY=30. Quick search on source did not reveal anything. 

I am running the Raspberry Pi version.

dividuum, 2016-03-16:
Thanks for the feedback. I'll reduce the output for video playback in the next version and add a way to configure the info interval.

dividuum, 2016-03-16:
For some reason I thought log level 1 was more verbose. When I set it to 1 (the default), all I get in the log output is:

     [glvideo.c] [0x1ca5400/vid/bla.mp4] video is 1280x720@30.000000

which doesn't seem verbose at all. Do you maybe use a different log level?

tuupola, 2016-03-16:
I get also this every 10 seconds. Info-Beamer is started via systemd (yeah Jessie) and all output goes to syslog. I know I could just ignore output and not log anything. Then again the output is useful, it just happens bit too often.

```
Mar 16 17:23:35 development-tv info-beamer:  uptime 12270s, cpu 1612s+985s, rss 21960kb, 1 objs, 70'C
Mar 16 17:23:35 development-tv info-beamer:  mem   fps  width height    cpu flag  name (alias)
Mar 16 17:23:35 development-tv info-beamer:  -----------------------------------------------------
Mar 16 17:23:35 development-tv info-beamer:  297kb  58.7   1920   1080   2.2%  R-s  '- content (-)
Mar 16 17:23:35 development-tv info-beamer:  ----------------------------------------------------
Mar 16 17:23:45 development-tv info-beamer:  uptime 12280s, cpu 1613s+986s, rss 22020kb, 1 objs, 70'C
```

dividuum, 2016-03-16:
A configurable interval for that output will be in the next release. I'll upload a new rerelease later today or tomorrow.

dividuum, 2016-03-17:
The newest prerelease (download here: https://info-beamer.com/download/player#prerelease) makes the interval for the info message configurable. By default it's 10 seconds but can be change by setting INFOBEAMER_INFO_INTERVAL=\<seconds\>.

Does that help?

tuupola, 2016-03-17:
Just tested and everything is perfect now. Thanks!

closed 2016-03-17 by tuupola