dividuum/info-beamer #35: Build process didn't work out of the box

stkdiretto, 2016-04-16:
I tried building info-beamer from source, which kind of failed – upon manually compiling glfw3, I had to change the LDFLAGS to the following:

`LDFLAGS += $(LUA_LDFLAGS) -llua5.1 -levent -lGL -lGLU -lglfw3 -lGLEW -lftgl -lIL -lILU -lavformat -lavcodec -lavutil -lswscale -lz -lm -ldl -lXi -lXinerama -lX11 -lXcursor -lXxf86vm -lXrandr -lpthread`

Did I do anything out of the ordinary here?

dividuum, 2016-04-18:
Probably not. I'm using Ubuntu 14.04 with https://launchpad.net/~keithw/+archive/ubuntu/glfw3 were everything works out of the box, so I haven't seen that problem yet. I'm not entirely sure why you need to explicitly link with -lXinerama -ldl -lXcursor. Can you provide the linker error message?

stkdiretto, 2016-04-18:
Original:
`cc -o info-beamer main.o image.o font.o video.o shader.o vnc.o framebuffer.o misc.o struct.o -L/usr/lib -llua5.1 -levent -lglfw -lGL -lGLU -lGLEW -lftgl -lIL -lILU -lavformat -lavcodec -lavutil -lswscale -lz -lm -lXi -lX11 -lXxf86vm -lXrandr -lpthread 
main.o: In Funktion `main':
main.c:(.text.startup+0x28f): Nicht definierter Verweis auf `glfwGetPrimaryMonitor'
main.c:(.text.startup+0x2b0): Nicht definierter Verweis auf `glfwCreateWindow'
main.c:(.text.startup+0x2cd): Nicht definierter Verweis auf `glfwSetFramebufferSizeCallback'
main.c:(.text.startup+0x2ea): Nicht definierter Verweis auf `glfwMakeContextCurrent'
main.c:(.text.startup+0x30a): Nicht definierter Verweis auf `glfwGetFramebufferSize'
main.c:(.text.startup+0x574): Nicht definierter Verweis auf `glfwWindowShouldClose'
main.c:(.text.startup+0x5d5): Nicht definierter Verweis auf `glfwSetInputMode'
collect2: error: ld returned 1 exit status
make: *** [info-beamer] Fehler 1`

dividuum, 2016-04-18:
I see. Where does `-Xinerama` and `-Xcursor` come from? Another error message once you add `-ld`? What's the output of `pkg-config --libs glfw3`?

stkdiretto, 2016-04-18:
Yep, the requirements keep on coming ;)

Example:
```
cc -o info-beamer main.o image.o font.o video.o shader.o vnc.o framebuffer.o misc.o struct.o -L/usr/lib -llua5.1 -levent -lglfw3 -lGL -lGLU -lGLEW -lftgl -lIL -lILU -lavformat -lavcodec -lavutil -lswscale -lz -lm -lXi -lX11 -lXxf86vm -lXrandr -lpthread -ldl
//usr/local/lib/libglfw3.a(x11_init.c.o): In Funktion `initExtensions':
x11_init.c:(.text+0x1b1b): Nicht definierter Verweis auf `XineramaQueryExtension'
x11_init.c:(.text+0x1b35): Nicht definierter Verweis auf `XineramaIsActive'
//usr/local/lib/libglfw3.a(x11_init.c.o): In Funktion `_glfwCreateCursor':
x11_init.c:(.text+0x2229): Nicht definierter Verweis auf `XcursorImageCreate'
x11_init.c:(.text+0x2300): Nicht definierter Verweis auf `XcursorImageLoadCursor'
x11_init.c:(.text+0x2310): Nicht definierter Verweis auf `XcursorImageDestroy'
//usr/local/lib/libglfw3.a(x11_monitor.c.o): In Funktion `_glfwPlatformGetMonitors':
x11_monitor.c:(.text+0x6fb): Nicht definierter Verweis auf `XineramaQueryScreens'
collect2: error: ld returned 1 exit status
```

```
$ pkg-config --libs glfw3
-L/usr/local/lib -lglfw3
$ pkg-config --print-requires --print-requires-private glfw3
x11
xrandr
xinerama
xi
xxf86vm
xcursor
gl
```

dividuum, 2016-04-18:
Interesting. Xinerama and xcursor are not requested by my version of glfw3.

    $ pkg-config --print-requires --print-requires-private glfw3
    x11
    xrandr
    xi
    xxf86vm
    gl

I guess I'll add them to the Makefile anyway as they are probably install everywhere.