dividuum/info-beamer #36: using layer on raspberry?

ozett, 2016-04-25:
hi,
i like info-beamer and started playing with it. 
i have omx-player displaying my webcams (which has runtime option for layers), 
and let infobeamer display the samples clock over it.

now i wanted to play with layer, but did not found more doku or hints on the website.

**is there an option to chose a certain layer for infobeamer?
can i have two instances on different layer?**

[your blog-post](https://info-beamer.com/blog/raspberry-pi-hardware-video-scaler) annouced some more information with videos on different layer,  but it seems you never coded/posted more of this?

> In a future blog post (follow @infobeamer on twitter to get notified when it's out) I'll talk about how to render videos into their own layer. 

it would be thankful for some hints to investigate the layer possibilties of this wonderful software...
tanks, ozett

ozett, 2016-04-25:
i gave two instances a try: it works
(at least for me with samples/clock and samples/hello)

it found, it would be more convenient to give the options (like different ports/ip/background) not from environment variables than the usual way with command-line on the specific instance...
but at least it worked. 
now it would be wonderful to control the layer number (per instance and in general..)
:smiley:

dividuum, 2016-04-25:
Just use

    INFOBEAMER_TARGET_L=<layer>

(see info-beamer -h for a complete list of parameters)

ozett, 2016-04-25:
cool, thanks a lot.
may you want to add an alias to the command-line options for the linux-typical
--help
-help
--?
-?

wich i tried without succes. me noob :smile: 

ever tried ffmpeg/socat or the like on linux...? will get you long listing...

`olaf@debian82:~$ socat --help
2016/04/25 13:49:51 socat[29585] E unknown option "--help"; use option "-h" for help
olaf@debian82:~$`

[image]

dividuum, 2016-04-25:
0.9.5 final will have more hints on how to reach the help. Thanks for bringing that up.

With regards to environment variables instead of command line arguments: I've found them easier to work with most of the time. In any real production environment, info-beamer is always started from some kind of process supervision framework (daemontools, systemd, ...) and running a Bash script that sets up the info-beamer runtime and configuration using environment variables is easier IMHO. Example of how that might look like:
    
    #!/bin/sh

    if jq -e .outside_sources < config.json > /dev/null; then
        export INFOBEAMER_OUTSIDE_SOURCES=1
    fi

    export INFOBEAMER_LOG_LEVEL=3
    export INFOBEAMER_BLANK_MODE=console

    exec nice -n -5 info-beamer /some/path

closed 2016-04-25 by dividuum