dividuum/info-beamer #37: font:write with RGB/texturelike. how to adress image-file?

ozett, 2016-04-25:
i am a noob at coding, 
but the syntax does not differentiate between RGB-numbers and picture-objects.
surely my mistake,
 but not superclear how to adress image-objects 
with the same font:write where RGB numbers are expected. 

can anybody help me out?

[image]

dividuum, 2016-04-25:
What you tried is correct, except that "texturelike" objects aren't supported on the Pi version, as it uses a completely different way of rendering text (that's a lot faster, see https://info-beamer.com/blog/performance-insights) which is incompatible with using a custom texture.

ozett, 2016-04-25:
ah, cool. 
thanks, 
that drove me deeper into the samples and working examples and the doku,
makes info-beamer more interesting than it was ...

cool would be showing rtsp/webcam stream insteand video-file, all with effects...
:smile: 

may you point this out in the wiki?

dividuum, 2016-04-26:
You might try loading external video streams. Just define the environment variable INFOBEAMER_OUTSIDE_SOURCES=1, then give resource.load_video an url. This feature isn't officially supported, as I don't have any experience with video streaming. But you can give it a try.

closed 2016-04-26 by dividuum

ozett, 2016-04-26:
i used the evironment variable, but:
"bad argument #1 to 'load_video' (invalid resource name. must be local file)"

[image]

dividuum, 2016-04-26:
You didn't set the environment variable correctly. I can tell, since info-beamer will put up a warning in the console output.

     $ INFOBEAMER_OUTSIDE_SOURCES=1 info-beamer /path/to/node

ozett, 2016-04-26:
thanks for dont letting me alone.... :smile: 
you are right. works perfectly...

i did the mistake and set the environment variables in the script with the normal user, than executed the info-beamer as root. i think, that was my mistake?
as i set the variable like you showed, in one line, it worked like a charm ...

WRONG:
[image]

dividuum, 2016-04-26:
Ok, good. Environment variables don't get set when running sudo, as that would be really dangerous: You could set a modified PATH which would probably lead to all kinds of problems if the called program doesn't handle that case.