dividuum/info-beamer #38: layers for nodes? can i adress or change layer-level on raspberry-pi ?

ozett, 2016-04-29:
hi,
i am not good in coding, but playing basic functions with info-beamer.
i love it! :smile: 

first i wanted to beg for a cycling-function in the videoplaylist-sample.
than i found the room-cycler fragment, copied nodes into it and now i can cycle 
between my RTSP webcam screens. WOW.
i played with background and transparancy. really cool.
with omxplayer i had with 192MB GPU split-screen-setting max 8 windows/streams. 
i will come to a test with info-beamer, if he beats this. 

BUT in the meantime my question: 
can i set layer-level for different nodes?

what i want to achieve is playing RTSP streams on different layer with transparency, pop them up and vanish them on different layer.

may help me roughly going this direction? or layers are not adressable "inside" info-beamer nodes?

dividuum, 2016-04-29:
If you use the normal way of playing video you don't need layers. The videos are rendered all within the same OpenGL layer. The order in which you :draw() them determines which video is drawn over each other.

Normally everything is inside a single OpenGL dispmanx (https://info-beamer.com/blog/raspberry-pi-hardware-video-scaler) layer. If you need better video performance it's possible to render videos outside of OpenGL. By default everything is in a single layer.

ozett, 2016-04-30:
can i change the order (means layer up/down) _while_ playing a video?
or do i have to stop/start the videop programmatically to push it on another Layer/level to move it in front or back..?
(my idea is to play with transparancey and layer/level overblending..)

dividuum, 2016-04-30:
As I said. There are no layers. Just draw video2 above video1, if you want video2 to be in front of video1. Reverse the draw order if you want video1 to be in front.

ozett, 2016-04-30:
ok. thanks for clarifying