dividuum/info-beamer #39: playing RTSP stream -- with buffer possible?

ozett, 2016-04-30:
thanks for helping me starting...
still love info-beamer, now getting videos from my rtsp-cams to work in overlay as child-nodes..
if i do this with more than one videostream, i am starting to get missing packets..

i did this before with omxplayer, it was possible to give a cache-option (" --video_queue 10") and i was able to get 8 windows/streams together with only some small lack/missing packets..

> omxplayer --layer -2  --adev local -n 1 --win '15 50 367 248' --crop '200 10 600 500' --stats --vol 0 --video_queue 10 --timeout 30 rtsp://192.168.14.105:7070/track2 

**is there also a buffer option for info-beamer and video.resource as RTSP?**
may you consider building one?  :smile: 

or i am going architectural totally wrong (as i am still a noob..) :smile: 

[image]

dividuum, 2016-04-30:
I think the idea is fine, but live streaming isn't a supported feature of info-beamer right now as I have pretty much no experience with that, so I don't feel confident to support it. So there won't be any related improvements in the near future.

ozett, 2016-04-30:
may you want to got this rtsp-way, very cooool with info-beamer...
(i cycle with the room-demo through nodes ... :thumbsup:
i cannot take screenshots with fbgrab of running info-beamer, so i photographed my 7inch display it as an appetizer. maybe..)  :yum:

[image]

mattenausohz, 2017-09-14:
Could anyone please send me the node.lua with a working rtsp usage? I cannot get it to work..

dividuum, 2017-09-14:
Set the environment variable `INFOBEAMER_OUTSIDE_SOURCES=1` and pray :-)

ozett, 2017-09-15:
one of my node.lua looks like:

```
-- gl.setup(1920, 1080)
gl.setup(800, 480)

local video
    -- video:dispose()
    --video = util.videoplayer("rtsp://192.168.14.104:554/ch0_1.h264")
    video = util.videoplayer("rtsp://192.168.14.98/4")

function node.render()

    --gl.rotate(0,0,0,0)
    --gl.clear(1,1,1,1)

    --video = resource.load_video("rtsp://192.168.14.103:554/ch0_1.h264")
    --video:draw( 0, 0, WIDTH, HEIGHT, 0.9)
    video:draw( 0, 0, WIDTH, HEIGHT, 1)
end
```

mattenausohz, 2017-09-15:
Thanks! Unfortunately I am getting many "RTS: missing packets" which leads to lots of artifacts in the stream... any idea how to improve this?

dividuum, 2017-09-15:
You might try the undocumented `INFOBEAMER_AV_DEBUG=1` setting with the 0.9.8 release. It will probably flood the output with debug messages, but maybe there's something useful in there?

mattenausohz, 2017-09-15:
This is now the debug using even a different camera (AXIS M1114). Any ideas?

[fileref.c] Streaming remote content isn't officially supported at the moment.
[fileref.c] All problems resulting from using this feature are your own.
[main.c] initialization completed
[rtsp @ 0x6bd00480] SDP:
v=0
o=- 1505466630621001 1505466630621001 IN IP4 10.160.37.12
s=Media Presentation
e=NONE
b=AS:50000
t=0 0
a=control:rtsp://10.160.37.12:554/axis-media/media.amp?videocodec=h264&streamprofile=low
a=range:npt=0.000000-
m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:50000
a=framerate:30.0
a=transform:1,0,0;0,1,0;0,0,1
a=control:rtsp://10.160.37.12:554/axis-media/media.amp/trackID=1?videocodec=h264&streamprofile=low
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1; profile-level-id=420029; sprop-parameter-sets=Z0IAKeKQCgC3YC3AQEBpB4kRUA==,aM48gA==

[rtsp @ 0x6bd00480] video codec set to: h264
[rtsp @ 0x6bd00480] RTP Packetization Mode: 1
[rtsp @ 0x6bd00480] RTP Profile IDC: 42 Profile IOP: 0 Level: 29
[rtsp @ 0x6bd00480] Extradata set to 0x6bd030e0 (size: 31)!
[rtsp @ 0x6bd00480] hello state=0
[h264 @ 0x6bd02bc0] RTP: missed 14 packets
[h264 @ 0x6bd02bc0] mb_type 26 in I slice too large at 17 24
[h264 @ 0x6bd02bc0] error while decoding MB 17 24
[h264 @ 0x6bd02bc0] concealing 1712 DC, 1712 AC, 1712 MV errors
[h264 @ 0x6bd02bc0] RTP: missed 69 packets
[h264 @ 0x6bd02bc0] corrupted macroblock 2 13 (total_coeff=-1)
[h264 @ 0x6bd02bc0] error while decoding MB 2 13
[h264 @ 0x6bd02bc0] concealing 2607 DC, 2607 AC, 2607 MV errors
[h264 @ 0x6bd02bc0] Frame num gap 6 4
[h264 @ 0x6bd02bc0] RTP: missed 35 packets
[h264 @ 0x6bd02bc0] corrupted macroblock 62 11 (total_coeff=-1)
[h264 @ 0x6bd02bc0] error while decoding MB 62 11
[h264 @ 0x6bd02bc0] concealing 2707 DC, 2707 AC, 2707 MV errors
[h264 @ 0x6bd02bc0] RTP: missed 1 packets
[h264 @ 0x6bd02bc0] RTP: missed 204 packets
[h264 @ 0x6bd02bc0] RTP: missed 7 packets
[h264 @ 0x6bd02bc0] RTP: missed 205 packets
[rtsp @ 0x6bd00480] All info found
[rtsp @ 0x6bd00480] Estimating duration from bitrate, this may be inaccurate
[glvideo.c] [0x20e2fe8/rtsp://xxx:xxx=7@10.160.37.12:554/axis-media/media.amp?videocodec=h264&streamprofile=low] no audio stream found
[glvideo.c] [0x20e2fe8/rtsp://xxx:xxx=7@10.160.37.12:554/axis-media/media.amp?videocodec=h264&streamprofile=low] video is 1280x720@0.000000
[NULL @ 0x6bd02bc0] RTP: missed 47 packets
[NULL @ 0x6bd02bc0] RTP: missed 38 packets
[NULL @ 0x6bd02bc0] RTP: missed 44 packets
[NULL @ 0x6bd02bc0] RTP: missed 15 packets
[NULL @ 0x6bd02bc0] RTP: missed 37 packets
[NULL @ 0x6bd02bc0] RTP: missed 37 packets

uptime 10s, cpu 1s+0s, rss 15100kb, 1 objs, 55'C
   mem   fps  width height    cpu flags  name (alias)
-----------------------------------------------------
 147kb  46.8   1920   1080   1.0%  R-s-  '- cue (-)
 115kb   0.0      0      0   0.0%  --s-     '- ola (-)
----------------------------------------------------

[NULL @ 0x6bd02bc0] RTP: missed 55 packets
[NULL @ 0x6bd02bc0] RTP: missed 2 packets
[NULL @ 0x6bd02bc0] RTP: missed 37 packets

ozett, 2017-09-15:
do you run infobeamer on a raspberry?
may you try the stream in omxplayer?

does the missing packets show up in the image on the screen?

mattenausohz, 2017-09-15:
Raspberry 3
All on cables
Will try on omxplayer
Same error on first cam (cheap one, thats why I tried the Axis)
The missing packets lead to an only half filled screen where the second half is filled with the last correct line repeated until the end.

ozett, 2017-09-16:
just one guess: i have a small screen setup,
maybe you try to test a small image rtsp-stream, like 100x100 ?
than check to increase image size and check debug for missing packets ?

what about the rasperry gpu-settings in raspi-config?
i increased this a lot ...

only guessing...maybe worth checking ..

mattenausohz, 2017-09-16:
If you read the first text here for this topic: it is exactly my problem. In omxplayer is seems to be possible to give a cache-option which helps. There is none of this for info-beamer obviously.

ozett, 2017-09-17:
ok, old thread, long ago my first entry.
i did not read it properly again - for analysing your problem.

now that i have it running my setup over years, i dont know anymore if there are missing packets.
it still works. i only see the stream-image. without glitches
no console anymore to check warnings. maybe there a still there... 
(i have to check explicitly now)

but, i dont see any drop outs in the image. 
(maybe on the console there are still missing packets warnings)

But, also, maybe i fiddelt with the GPU settings in the raspberry to improve my first error.
did you updgraded the whole firmware/raspian to the latest?
i can remember to have done this in the past to solve some of infobeamer errors..

maybe also worth checking this for you?
https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md

dividuum, 2017-09-18:
If you find out how/if the omxplayer cache option maps to an `AVDictionary` setting that gets passed into 
`avformat_open_input`, we might give that a try. My cargo cult senses tell me that options like this should
end up there. :-)

ozett, 2017-09-19:
result from running the rtsp-stream with omxplayer?

mattenausohz, 2017-10-04:
Sorry, I am still on but a bit busy.. will check and come back

stevemenke, 2018-01-27:
Any updates on this. I am having the same problem with dropped packets when streaming from an IP camera. It results in a lost of vertical streaks on the screen.

dividuum, 2018-01-29:
No update from my side. If anyone provide an easy way to me to locally reproduce this problem, I'd be happy to help. Otherwise I'm a bit lost on how to approach solving this.

uzlonewolf, 2019-04-08:
I use udp, but it's set the same way; just add "buffer_size" to the URL query string.  My go-to is ```udp://239.255.42.46:5004?buffer_size=5000000&overrun_nonfatal=1&fifo_size=100000 live=1```  (yes, the space before "live=1" is correct)

If you see ```[udp @ 0x68a00a70] attempted to set receive buffer to size 5000000 but it only ended up set as 327680``` you need to ```echo 4194304 > /proc/sys/net/core/rmem_max```. 

```gl.setup(1920, 1080)

local video
    video = util.videoplayer("udp://239.255.42.46:5004?buffer_size=5000000&overrun_nonfatal=1&fifo_size=100000 live=1")

function node.render()
    video:draw( 0, 0, WIDTH, HEIGHT, 1)
end
```

Works great :)

ozett, 2019-04-08:
...one year later....
but is the buffer-option in the URL specific to your camera?
or a general option for info-beamer?

seems to me, that this url-buffer-option only works with cameras, which have that option for rtsp-streams (even udp) ??

dividuum, 2019-04-08:
That's [udp protocol options](https://ffmpeg.org/ffmpeg-protocols.html#udp). Handled by ffmpeg/libav which info-beamer uses.

ozett, 2019-04-08:
cool! 
thanks for clarification. 

i giv this a rocket too. cool option for rtsp-improvement.
👍

uzlonewolf, 2019-04-08:
Looking at the docs it looks like upping reorder_queue_size may help as well; I'd see what ```?reorder_queue_size=1000&buffer_size=5000000``` does.