dividuum/info-beamer #40: Question/Informational: child nodes allocation permanent ressources?

ozett, 2016-05-05:
hi,
i am using the room-cycling example ([from here](https://github.com/dividuum/info-beamer-nodes)) and added two nodes, playing a video in a loop.
i am not aware, if the looping video blocks permanent ressources, it the room-cycling.lua switches the nodes. to put it the other way: are the nodes allocating resources on my raspberry, even if i dont see them? or only, if they are switched on?

i ask this, because i considered as a next step to have more nodes with playing video-files and cylcle/switch trough them on my raspberry. 

#### i observed somehow display-issues/performance lacks  
while playing 8 rtsp-streams simultaneously, while the same streams played fine with omxplayer. maybe this is another issue, but you could help me in advising how to cycle through videos with saving resouces. (may to release/kill them, but can i still switch to nodes with looping videos?

dividuum, 2016-05-06:
Yes. All nodes are consuming resources. Video that are not stopped continue playing, even if they are not displayed anywhere. If you want to avoid that, you have to explicitly :stop() and later :start() them. Additionally every loaded video, even if paused consumes GPU memory and other resources. I don't recommend opening more than 4-5 videos at once. While child nodes make things easier, they are not very good if performance is critical: Have a look at https://info-beamer.com/blog/introducing-nested-nodes

Here is example code that shows how to switch between multiple videos: https://github.com/dividuum/info-beamer-nodes/tree/master/videolist

ozett, 2016-05-06:
hey, thanks :smiley: 
i looked up your links and articles with great interest. 
i will give it a big try, as i am a little noob in coding,
but i guess your hints will help me a lot..
thanks again,
:+1: