dividuum/info-beamer #41: CRITICAL ERROR: need opengl 3.0 support

dennisthegitmenace, 2016-05-31:
I installed the open source edition on Ubuntu 14.04 (although did have troubles with dependencies installation).

When i run ` ./info-beamer samples/hello`
I get the following output:
```
Info Beamer 1.0pre4.47a5ea+Lua 5.1 (http://info-beamer.org/)
Copyright (c) 2015 Florian Wesch <fw@dividuum.de>

[main.c] chdir /home/denys/info-beamer/samples
[main.c] root node is hello
[main.c] tcp/udp port is 4444
[main.c] initial size is 1024x768
CRITICAL ERROR: need opengl 3.0 support
```
Any suggestions on how to fix it ?

dividuum, 2016-05-31:
This means that your graphics driver is most likely not capable of running info-beamer. That's the output of

```
glxinfo |egrep "OpenGL|glx"
```

Are you running info-beamer in a virtual machine? If so, #22 is probably relevant.

dennisthegitmenace, 2016-05-31:
Yep, it's a virtual machine. Thank you. I'l try to figure it out and let you know how it went.