dividuum/info-beamer #43: animated gif supported?

ozett, 2016-07-03:
i would be in heaven, if animated gifs are supported. is this the case?

the docu https://info-beamer.com/doc/info-beamer#resource.load_image says not: " info-beamer supports loading JPEG, PNG and on the PI version PKM images." but this issue https://github.com/dividuum/info-beamer/issues/15 seems to deal with .gifs. (maybe not animated ones :question: )

i found a python lib https://github.com/chidea/FBpyGIF, which shows nicely all animated gifs in autozoom on my 1080x1920 display (wich is 90 degree rotated on my raspberry9, but info-beamer has some more advantages.. :smile: 

i would be in heaven, if animated gifs could be displayed...may in a beta? :smiling_imp:

dividuum, 2016-07-13:
The version in the repository uses a different image loading library ([DevIL](http://openil.sourceforge.net/)) compared to the raspberry pi version, which uses custom loading routines. So this version supports loading gifs (but only the first frame) while the Pi version refuses to load anything except for png/jpg/pkm.

In my opinion playing gifs feels like a waste of resources. Either you have to decompress all frames before playback or you decode in realtime. For small animation that will work, but for bigger animations that might be a waste of either memory or CPU compared to using h264 videos which can be hardware decoded on the Pi.

What exactly is your usecase for gifs? Also sorry about the delayed response, I've been away for the last 10 days and didn't have regular or stable internet access.

ozett, 2016-07-14:
hi, 
thanks for reply. 
i played with animated gifs through a python-lib directly on the framebuffer.
as i played before with info-beamer (which had a lot of more posibilties) i looked ideal to me, to have also the gif-animations played on layers and programmatically changing at one place: in info-beamer.
i did not now deeper about the inner workings, but i was at first a striking idea to also play animated phots (i.e. gifs) in info-beamer and have its astounding features for maniplulation/programming.
i build an example with both together (python-lib behind info-beamer) for testing here:
https://cloud.githubusercontent.com/assets/5937811/16357642/a789b414-3afc-11e6-8ec3-0214d4699467.png and played with it. (https://github.com/chidea/FBpyGIF/issues/1)

the more features, the more ideas will arise in software, maybe it is an option for you to build support in? at the moment i did not noticed any performance problems, but i will look this way...

ozett, 2016-07-14:
one use case:  :smile: 

[image]

dividuum, 2016-07-14:
Have you tried using h264 videos instead? `ffmpeg -i some.gif -pix_fmt yuv420p out.mp4` usually does the trick (I think). If you need gif to transparency, I'm sure there's a way to somehow set a chroma key in ffmpeg and then use a shader to set the alpha value of that color to zero.

I'm fairly certain that I won't add gifs anytime soon unless development is sponsored.

ozett, 2016-07-14:
hmmmm,
i read your hint for conversion somewhere in a raspberry forum, but i think it is not really doable for me. i had to double all of my gif collection (which is running fine with python) and sits between some other pictures for the picture frame. as i would add/change, i had to reflect this with additional conversion. i could this with inotify-scripts, but at least it would double my items and my handling...
thanks for going into this issue, maybe there will be a library (by suprrise?), which will bring support for animated gifs without great effort.. 
i will stay tuned on this here ... :smiley:

davidecat, 2018-03-27:
> Have you tried using h264 videos instead? ffmpeg -i some.gif -pix_fmt yuv420p out.mp4 usually
> does the trick (I think). If you need gif to transparency, I'm sure there's a way to somehow set a
> chroma key in ffmpeg and then use a shader to set the alpha value of that color to zero. 
Hello! Im very interested about this feature. 
Where can i find an example of the code to create the shader with the h264 video and set the green opacity (0,255,0) to alpha 0? 

Thank you!