dividuum/info-beamer #44: no sound in videos

MickeyO-N3il, 2016-08-15:
Hi,

Info-Beamer is not playing any sound, if i start a video. But if I play the same video with the omxplayer everything works fine.

The environment variable is set to: INFOBEAMER_AUDIO_TARGET=hdmi
Audio Output is set to HDMI: `amixer cset numid=3 2`

And this is my very short script:
```
gl.setup(NATIVE_WIDTH, NATIVE_HEIGHT)

local video = resource.load_video{
    file = "VID_20160815.mp4";
    audio = true;
    looped = false;
    paused = false;
}

function node.render()
    video:draw(0, 0, WIDTH, HEIGHT)
end
```

I'm using Info-Beamer Pi on RASPBIAN JESSIE (Raspberry Pi B).

Any idea what is going wrong? I'm quite new to Linux and I'm thankful for any help.

Cheers
Mickey

dividuum, 2016-08-15:
On first sight that looks fine. Could you send me the output of running

```
$ INFOBEAMER_AUDIO_TARGET=hdmi INFOBEAMER_LOG_LEVEL=3 info-beamer /some/path
```

Either here or to the mail address on https://info-beamer.com/doc/about

MickeyO-N3il, 2016-08-16:
It works with the `INFOBEAMER_AUDIO_TARGET=hdmi` parameter! So i looked in `printenv` and the environment variable was gone. 
If I write `export INFOBEAMER_AUDIO_TARGET=hdmi` the variable is only working till the next reboot.

dividuum, 2016-08-16:
When you have a shell running, setting environment variables is temporary. By default they are not inherited by programs you call from your shell. You have to either
 
 * export then like this `$ export VARIABLE=value`. After that, programs called from your shell will see that value.
 * call the program with the variable assignment like this `VARIABLE=value program`

In both cases once you close the shell the assignment is gone. My suggestion would be to write a small wrapper script that sets the correct values. Something like this:

```
#!/bin/sh
export INFOBEAMER_AUDIO_TARGET=hdmi
exec info-beamer /some/path
```

closed 2016-08-16 by dividuum

MickeyO-N3il, 2016-08-16:
Ok, Thanks. Didn't knew that.
I will add the command to the other script, which is calling info-beamer.