dividuum/info-beamer #46: Get Raspberry Pi IP address

RajshekharHoratti, 2016-10-18:
Hello, I am new to lua programming and i was trying to get my raspberry pi's ip address from lua i tried using 

`os.execute('ifconfig wlan0 | grep "inet\ addr" | cut -d: -f2 | cut -d" " -f1')`
but getting  an error.

> invalid escape sequence near ''ifconfig wlan0 | grep "inet'

So is there any other way to get ip address from lua.

dividuum, 2016-10-18:
What exactly are you trying to achieve? From the Lua runtime within info-beamer `os.execute` isn't available. My suggestion would be to either

 * Write the IP address into a file using something like your command and then read that file with [resource.load_file](https://info-beamer.com/doc/info-beamer#resource.load_file)

 * Export an environment variable `INFOBEAMER_ENV_IP=1.2.3.4` and then use `sys.get_env "IP"` ([documentation](https://info-beamer.com/doc/info-beamer#sys.get_env)) in Lua.

RajshekharHoratti, 2016-10-20:
Thanks a lot it worked..!! :)

closed 2016-10-20 by RajshekharHoratti