dividuum/info-beamer #47: Is it possible to set different display duration for Different images

Philastone, 2016-10-19:
Hi, i am still new to info-beamer. I am trying to display 5 images in one folder for 20 seconds each and that works fine. Now I want to display 2 of the images for 30 second each and the rest should be displayed for 10 seconds. 

Is there a way to do this?

Just a thought I had if the above is not possible, is it also possible to run nested node.lua scripts? By nested, i mean having a set of images with a lua script on a main folder and then have a sub folder with a second set of images also with their own lua script?

Your help will greatly be appreciated

dividuum, 2016-10-19:
Sure, that's possible. This [code](https://github.com/info-beamer/package-hd-player), for example, does that. Configuring it is a bit complicated as you have to create a config.json file. In your case it might look like this:

```
{
  "playlist": [
    {
      "duration": 5,
      "file": {
        "asset_name": "video.mp4",
        "type": "video"
      }
    },
    {
      "duration": 5,
      "file": {
        "asset_name": "image.jpg",
        "type": "image"
      }
    }
  ],
  "switch_time": 0,
  "kenburns": false,
  "synced": true,
  "progress": "no",
  "rotation": 90,
  "audio": true
}
```

So put all your files into the same directory, create a `config.json` with the above content (and feel free to add more videos/images), then run info-beamer in that directory.

Philastone, 2016-10-20:
Thanks alot for this. 

I just have a lil confusion... Do i just create a folder with my images/videos and the config.json and no node.lua file? or do I need to call the config.json somehow on my node.lua file?

dividuum, 2016-10-20:
The code I linked includes a `node.lua` file. You put all those files together with your images/videos and the `config.json` into a single directory. Once info-beamer is started it will load `node.lua` which will then use `config.json` to decide which content to show.

Philastone, 2016-10-20:
Thanks

Sadly i am not winning. I have done as you have instructed but i am getting a runtime error and I cannot figure out why is it throwing the error. 

The error looks like this :
`[phila] runtime error: phila/node.lua:122: Expected value but found invalid token at character 131
stack traceback:
	[C]: in function 'decode'
	phila/node.lua:122: in function 'handler'
	bundled userlib.lua:303: in function 'updated'
	bundled userlib.lua:309: in function 'file_watch'
	phila/node.lua:120: in function <phila/node.lua:110>
	phila/node.lua:176: in main chunk
	kernel: in function 'run_in_sandbox'
[main.c] initialization completed
`

I have tried tracing through the code but I have had no luck. I am using exactly the same code from the example with an addition of the config.json file.
Your help will be greatly appreciated

dividuum, 2016-10-20:
This means that you have a syntax error in your json file. Make sure it's correct. You might use the `jq` tools (you probably have to install that) for that:

Example:
```
$ jq config.json
parse error: Expected separator between values at line 10, column 5
```