dividuum/info-beamer #49: Play all Video's in a Json file in loop

RajshekharHoratti, 2016-11-23:
Hi is there any way to play all the videos which are in the json file in a loop for example :- i'm having a json file like this 
`[{"state": "active", "video":"video1.mp4"},{"state": "active", "video":"video2.mp4"},{"state": "inactive", "video":"video3.mp4"}]`

i want to play all the videos who's `"state" = "active"`
and Thanks :)

dividuum, 2016-11-23:
Have a look at https://github.com/dividuum/info-beamer-nodes/tree/master/videolist. It's almost what you need. You just have to replace the code in util.file_watch to parse your json file instead. You can use the builtin [json module](https://info-beamer.com/doc/info-beamer#util.file_watch) for that.

RajshekharHoratti, 2016-11-26:
Thanks for the reply it exactly what i needed thanks a lot :)

closed 2016-11-26 by RajshekharHoratti