dividuum/info-beamer #51: Build Error on ubuntu 16.10

PowerPan, 2017-01-13:
On A Fresh Ubuntu Install 16.10 following the build instructions from [buildingfromsource](https://info-beamer.com/doc/info-beamer#buildingfromsource) i  get this error:

```
johannes@ubuntu ~/info-beamer (git)-[master] % make                                                                                                                                                    
cc -O3 -DNDEBUG -DVERSION='"1.0pre4.47a5ea"' -I/usr/include/lua5.1 -I/usr/include/freetype2/ -I/usr/include/ffmpeg -std=c99 -Wall   -c -o video.o video.c
video.c: In function ‘video_open’:
video.c:81:21: error: ‘PIX_FMT_RGB24’ undeclared (first use in this function)
     video->format = PIX_FMT_RGB24;
                     ^~~~~~~~~~~~~
video.c:81:21: note: each undeclared identifier is reported only once for each function it appears in
video.c:146:24: warning: implicit declaration of function ‘avcodec_alloc_frame’ [-Wimplicit-function-declaration]
     video->raw_frame = avcodec_alloc_frame();
                        ^~~~~~~~~~~~~~~~~~~
video.c:146:22: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     video->raw_frame = avcodec_alloc_frame();
                      ^
video.c:147:25: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
     video->scaled_frame = avcodec_alloc_frame();
                         ^
video.c:155:5: warning: ‘avpicture_get_size’ is deprecated [-Wdeprecated-declarations]
     video->buffer = av_malloc(avpicture_get_size(
     ^~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4898:5: note: declared here
 int avpicture_get_size(enum AVPixelFormat pix_fmt, int width, int height);
     ^~~~~~~~~~~~~~~~~~
video.c:162:5: warning: ‘avpicture_fill’ is deprecated [-Wdeprecated-declarations]
     avpicture_fill(
     ^~~~~~~~~~~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4883:5: note: declared here
 int avpicture_fill(AVPicture *picture, const uint8_t *ptr,
     ^~~~~~~~~~~~~~
video.c: In function ‘video_next_frame’:
video.c:206:9: warning: ‘av_free_packet’ is deprecated [-Wdeprecated-declarations]
         av_free_packet(&packet);
         ^~~~~~~~~~~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4040:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
video.c:213:9: warning: ‘av_free_packet’ is deprecated [-Wdeprecated-declarations]
         av_free_packet(&packet);
         ^~~~~~~~~~~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4040:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
video.c:224:9: warning: ‘av_free_packet’ is deprecated [-Wdeprecated-declarations]
         av_free_packet(&packet);
         ^~~~~~~~~~~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4040:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
video.c:254:5: warning: ‘av_free_packet’ is deprecated [-Wdeprecated-declarations]
     av_free_packet(&packet);
     ^~~~~~~~~~~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4040:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
<eingebaut>: die Regel für Ziel „video.o“ scheiterte
make: *** [video.o] Fehler 1
```

dividuum, 2017-01-15:
I only have had a 16.10 running in a cloud vm somewhere, so I can't test if the result works, but I made those changes to video.c:

```
diff --git a/video.c b/video.c
index e58b009..b737b1d 100644
--- a/video.c
+++ b/video.c
@@ -78,7 +78,7 @@ static void video_free(video_t *video) {
 
 static int video_open(video_t *video, const char *filename) {
     video->finished = 0;
-    video->format = PIX_FMT_RGB24;
+    video->format = AV_PIX_FMT_RGB24;
 
     if (avformat_open_input(&video->format_context, filename, NULL, NULL) ||
             avformat_find_stream_info(video->format_context, NULL) < 0) {
@@ -143,8 +143,8 @@ static int video_open(video_t *video, const char *filename) {
     fprintf(stderr, INFO("fps: %lf\n"), video->fps);
 
     /* Get framebuffers */
-    video->raw_frame = avcodec_alloc_frame();
-    video->scaled_frame = avcodec_alloc_frame();
+    video->raw_frame = av_frame_alloc();
+    video->scaled_frame = av_frame_alloc();
 
     if (!video->raw_frame || !video->scaled_frame) {
         fprintf(stderr, ERROR("cannot preallocate frames\n"));
@@ -152,10 +152,11 @@ static int video_open(video_t *video, const char *filename) {
     }
 
     /* Create data buffer */
-    video->buffer = av_malloc(avpicture_get_size(
+    video->buffer = av_malloc(av_image_get_buffer_size(
         video->format, 
         video->buffer_width, 
-        video->buffer_height
+        video->buffer_height,
+        16
     ));
 
     /* Init buffers */
```

There are still some deprecation warnings that I'll have to eventually fix, but it should work. Can you give it a try?

PowerPan, 2017-01-16:
Ok after making those changes this is the patch for this (My Line Numbers where different than yours?!)

```
johannes@ubuntu ~/info-beamer (git)-[master] % make
cc -O3 -DNDEBUG -DVERSION='"1.0pre4.47a5ea"' -I/usr/include/lua5.1 -I/usr/include/freetype2/ -I/usr/include/ffmpeg -std=c99 -Wall   -c -o video.o video.c
video.c: In function ‘video_open’:
video.c:155:31: warning: implicit declaration of function ‘av_image_get_buffer_size’ [-Wimplicit-function-declaration]
     video->buffer = av_malloc(av_image_get_buffer_size(
                               ^~~~~~~~~~~~~~~~~~~~~~~~
video.c:163:5: warning: ‘avpicture_fill’ is deprecated [-Wdeprecated-declarations]
     avpicture_fill(
     ^~~~~~~~~~~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4883:5: note: declared here
 int avpicture_fill(AVPicture *picture, const uint8_t *ptr,
     ^~~~~~~~~~~~~~
video.c: In function ‘video_next_frame’:
video.c:207:9: warning: ‘av_free_packet’ is deprecated [-Wdeprecated-declarations]
         av_free_packet(&packet);
         ^~~~~~~~~~~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4040:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
video.c:214:9: warning: ‘av_free_packet’ is deprecated [-Wdeprecated-declarations]
         av_free_packet(&packet);
         ^~~~~~~~~~~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4040:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
video.c:225:9: warning: ‘av_free_packet’ is deprecated [-Wdeprecated-declarations]
         av_free_packet(&packet);
         ^~~~~~~~~~~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4040:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
video.c:255:5: warning: ‘av_free_packet’ is deprecated [-Wdeprecated-declarations]
     av_free_packet(&packet);
     ^~~~~~~~~~~~~~
In file included from video.c:35:0:
/usr/include/x86_64-linux-gnu/libavcodec/avcodec.h:4040:6: note: declared here
 void av_free_packet(AVPacket *pkt);
      ^~~~~~~~~~~~~~
cc -O3 -DNDEBUG -DVERSION='"1.0pre4.47a5ea"' -I/usr/include/lua5.1 -I/usr/include/freetype2/ -I/usr/include/ffmpeg -std=c99 -Wall   -c -o shader.o shader.c
cc -O3 -DNDEBUG -DVERSION='"1.0pre4.47a5ea"' -I/usr/include/lua5.1 -I/usr/include/freetype2/ -I/usr/include/ffmpeg -std=c99 -Wall   -c -o vnc.o vnc.c
In file included from /usr/include/assert.h:35:0,
                 from vnc.c:4:
/usr/include/features.h:148:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
cc -O3 -DNDEBUG -DVERSION='"1.0pre4.47a5ea"' -I/usr/include/lua5.1 -I/usr/include/freetype2/ -I/usr/include/ffmpeg -std=c99 -Wall   -c -o framebuffer.o framebuffer.c
cc -O3 -DNDEBUG -DVERSION='"1.0pre4.47a5ea"' -I/usr/include/lua5.1 -I/usr/include/freetype2/ -I/usr/include/ffmpeg -std=c99 -Wall   -c -o misc.o misc.c
cc -O3 -DNDEBUG -DVERSION='"1.0pre4.47a5ea"' -I/usr/include/lua5.1 -I/usr/include/freetype2/ -I/usr/include/ffmpeg -std=c99 -Wall   -c -o struct.o struct.c
cc -o info-beamer main.o image.o font.o video.o shader.o vnc.o framebuffer.o misc.o struct.o -L/usr/lib -llua5.1 -levent -lglfw -lGL -lGLU -lGLEW -lftgl -lIL -lILU -lavformat -lavcodec -lavutil -lswscale -lz -lm -ldl -lXi -lX11 -lXxf86vm -lXrandr -lXinerama -lXcursor -lpthread 
/usr/bin/ld: cannot find -lXinerama
/usr/bin/ld: cannot find -lXcursor
collect2: error: ld returned 1 exit status
Makefile:43: die Regel für Ziel „info-beamer“ scheiterte
make: *** [info-beamer] Fehler 1
```

So i do:
```
sudo aptitude install libxcursor-dev libxinerama-dev
```

After that i can make and make install

fenhl, 2017-07-15:
Running into the same error on Debian stretch. Applying the patch seems to fix the issue.