dividuum/info-beamer #54: video in root-node not working correct, better in child node. but doku says use root-node

ozett, 2017-02-06:
hi,
as i follow the dokumentation, i would stick to one root-node.

[image]

i try to display a videostream, but as i try this like 

```
video01 = resource.load_video("rtsp://192.168.14.98/4")
util.draw_correct(video01,10,10,362,298,1)
```

it shows getting rtsp-data, but no image on screen.

if i put this into a child-node and call the child from the root with,
```
util.draw_correct(resource.render_child("cam-hof2"),10,10,362,298,0.7)
```
it is working fine 😵 

the doku advises me to stay into root, 

[image]

but it runs better in child.  Some advise for best practice this?
🤝

dividuum, 2017-02-06:
Just to make sure: You don't literally use this code, right?

```Lua
video01 = resource.load_video("rtsp://192.168.14.98/4")
util.draw_correct(video01,10,10,362,298,1)
```

but something like:

```Lua
local video01 = resource.load_video("rtsp://192.168.14.98/4")
function node.render()
    util.draw_correct(video01,10,10,362,298,1)
end
```

ozett, 2017-02-06:
you hit me 👊 

as i am a coding illiterate, but love to see thinks happen (these stupids are sometimes called "makers") 😁 ,
i literally used the code like i showed it above. 😵 

you pointed me with the nose to my mistake 👃 , 
and i now defined the video resource _before_ the node.render(), 
means outside the main loop (i guess), ▶️ 
and as used your code architecture. boom, 💥  
it works...

but you ever tried my example? there is not really a syntax check, or somekind of logic-code-check.
nor really an error on the cmd-line.  i runs, on the cmd-line there is only error of getting the rtsp-stream...
and it stucks after i while. i now think i now, why... 😁