dividuum/info-beamer #55: how works the skaling of childs or utils-resource with gl.setup values correct?

ozett, 2017-02-06:
hi,
i have different unpredicable results. may you can clear somethind up/help me with this ❔ 

i want to display a 352x288 video-resource.
my native screen is portrait-mode 480x800 on an raspberry.

i set up a child-node with the video-resouce (because child works best), and made this video-sizes in there:
```
gl.setup(352, 288)

local video = util.videoplayer("rtsp://192.168.14.98/4")

function node.render()
    video:draw( 0, 0, 352, 288, 1)
end
```

now i call the child-node from the root-node with:

```
gl.setup(NATIVE_WIDTH, NATIVE_HEIGHT)
--gl.setup(480, 800)
...
util.draw_correct(resource.render_child("cam-hof"),30,10,362,298,0.7)
```

as far as i understand, i would thought to control the position of the child with the coordinates i gave with calling the child. 
but i cannot manage to put it on the right side of the whole screen. 
it shrinks, it i increase the x-values.

also i dont understand/nor can i predict how info-beamer scales different values in calling the child-node and makes use of the settings given in the child. 

i only have (to me) strange scaling-results. may you an elaborate on this a bit?

[image]

how to do this best?
thanks a lot ...

dividuum, 2017-02-06:
> i want to display a 352x288 video-resource.
> my native screen is portrait-mode 480x800 on an raspberry.

Do you want to scale up the video to the full screen space or preserve its aspect ratio? In either case, using a child node is a waste of resources. Just render the video in the top level node. If you want to place it in a set square without any resizing to preserve the aspect ratio, just use:

```Lua
video:draw(x1, y1, x2, y2)
```

and the video will end up in that area. If you want to preserve the aspect ratio, use

```Lua
util.draw_correct(video, x1, y1, x2, y2)
```

and info-beamer will fit the video in the specified area while preserving its aspect ratio. This is always done by scaling the object up or down so it fits in.

> as far as i understand, i would thought to control the position of the child with the coordinates i gave with calling the child.

That is correct, unless you use util.draw_correct, which preserves the aspect ratio (by scaling things down until they fit in the specified area).

ozett, 2017-02-06:
> That is correct, unless you use util.draw_correct, which preserves the aspect ratio (by scaling things down until they fit in the specified area).

thanks for the information, i still ask myself what happens while preserving the aspect-ratio to the size of the video, if x2 and y2 are to narrow - or to wide, i.e. somehow errornes values for correct aspect-ratio..

wouldn't it be enough to preserve aspect-ratio to have only x2 or y2?

ozett, 2017-02-06:
and still i ask myself, in what priority info-beamer handles the given child-node dimensions, if the child-node itself is called with other dimensions. will everting be nested into each other, like russian-dolls?

dividuum, 2017-02-06:
> [..] what happens while preserving the aspect-ratio to the size of the video, if x2 and y2 are to narrow - or to wide, i.e. somehow errornes values for correct aspect-ratio..

It scales the source up or down, so that it fits. See https://github.com/dividuum/info-beamer/blob/47a5ea1b976a3cf90f97cfc1e7218fe89d093108/userlib.lua#L318

> [...] in what priority info-beamer handles the given child-node dimensions, if the child-node itself is called with other dimensions. will everting be nested into each other, like russian-dolls?

The is no magic. The child node sets its own size. The node calling `render_child` gets an image with that dimensions. It can then draw that in whatever size it wishes.

ozett, 2017-02-06:
> The node calling render_child gets an image with that dimensions. It can then draw that in whatever size it wishes.

i see. 
if within the child-node with its given dimension there is some smaller window defined (like my video), 
and than the child-node will called with its own scaling dimensions, then accordingly within the child-window the smaller video-window is scaled accordingly smaller ...

thanks for explaining this..

i also tested the aspect-ratio-scaling. if i call my video stream of 352x288
```
util.draw_correct(video01,100,10,462,298,1)
```
it is +100px on x-dimension in full size of the video, like i expected. and
```
util.draw_correct(video01,100,10,462,98,1)
```
scales it down to the dimension of the y2 value. preserving the aspect-ratio.

somekind of understandable logic...
if the y2 value has to be valid, the x2 value must be ignored to preserve aspect-ratio.
💡

dividuum, 2017-02-06:
I'm not sure I understand the question. There is no priority. The source (video01 in your case)  is placed into the area defined by x1, y1, x2, y2 in a way the its aspect ratio is preserved. See my previous link.

ozett, 2017-02-07:
maybe i am unclear to myself. lets think out loud...

i have a video of 352x288.
i call this resource with util.draw_correct to preserve aspect-ratio.
and give x1,y1 as left corner start point with 10,10
and give x2,y2 as right-corner end-point with 520,298.

what size will the video window be -- at which position?

i assume i will be drawn with left-corner startpoint x,y =10,10 and
as the destination-size of the window was given with x2,y2 of 520,298 
the real video-size of 352x288 (to preserve aspect-ratio) be sized to the side of the y-line, 
this will be, because the y-line with 298 pixel is the smaller value over the x-value of 520 px. 
❓ 

i believe somehow i am wrong, but where lies the mistake in my assumtions ?

dividuum, 2017-02-07:
```Lua
local x1 = 10
local y1 = 10
local x2 = 520
local y2 = 298

local ox1, oy1, ox2, oy2 = util.scale_into(x2 - x1, y2 - y1, 352, 288)
print(x1 + ox1, y1 + oy1, x1 + ox2, y1 + oy2)
```

=> `89      10      441     298`

Centered into the area given by x1, y1, x2, y2.

ozett, 2017-02-07:
> Centered

That solves the riddle! 😁 

thanks a lot for clearification!

closed 2017-02-07 by dividuum