dividuum/info-beamer #57: cant run on native ubuntu 16.04

x29a, 2017-02-15:
hi,

im trying to run info-beamer opensource on a native ubuntu server 16.04 machine. I read somewhere, that info-beamer can run without X, or is this only valid for the rpi version?

i downloaded the prerequisites according to [the docs](https://info-beamer.com/doc/info-beamer#prerequisitesonubuntu) and built the software with debug via

> DEBUG="1" make

im logged in via ssh as root, the output i get is

> root@host:/opt/info-beamer# ./info-beamer samples/
> Info Beamer 1.0pre4.47a5ea+Lua 5.1 (http://info-beamer.org/)
> Copyright (c) 2015 Florian Wesch <fw@dividuum.de>
> 
> [main.c] chdir /opt/info-beamer
> [main.c] root node is samples
> [main.c] tcp/udp port is 4444
> CRITICAL ERROR: cannot initialize glfw

with the lua version beeing
> root@host:/opt/info-beamer# lua -v
> Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio

and graphics hardware
> root@host:/opt/info-beamer# lspci | grep VGA
> 01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] RV516 [Radeon X1300/X1550 Series]

after applying pull request #56 i see
> glfw error 65544 (X11: The DISPLAY environment variable is missing)

so X is required?

dividuum, 2017-02-15:
You'll probably need X, as I'm unaware of any hardware accelerated OpenGL implementation that doesn't use X on the Desktop. I might be wrong.

I suggest you use X, start one of those fullscreen window managers and then info-beamer. That should be pretty minimal.

x29a, 2017-02-16:
Ok thanks, i installed XOrg, am starting the X server via rc.local and then info-beamer on top.

The openGL issue (similar to #41) is "fixed" by commenting the [opengl version check](https://github.com/dividuum/info-beamer/blob/master/main.c#L1566) out.

closed 2017-02-16 by x29a