dividuum/info-beamer #59: Segfault

fenhl, 2017-04-11:
I'm running the open-source version from within KDE on Debian stable, and it always exits with an exit code of 139 (SIGSEGV) after around 4 to 5 minutes, independently of the node being displayed.

dividuum, 2017-04-11:
Can you recompile with -ggdb and send me a gdb traceback?

fenhl, 2017-04-11:
How do I do that? So far I have run `make clean`, `make DEBUG=1`, and installed `gdb`. I can't find anything related to gdb tracebacks though.

dividuum, 2017-04-11:
You start info-beamer like this:

```
gdb info-beamer
```

Then in gdb you use

```
run /path/to/node
```

which starts info-beamer. Once it segfaults, you'll be back in gdb and you can type `bt` to get a backtrace. That will hopefully help me understand what's happening.

fenhl, 2017-04-11:
```
$ gdb /opt/git/github.com/dividuum/info-beamer/master/info-beamer
GNU gdb (Debian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/git/github.com/dividuum/info-beamer/master/info-beamer...done.
(gdb) r /home/fenhl/info-beamer-blank
Starting program: /opt/git/github.com/dividuum/info-beamer/master/info-beamer /home/fenhl/info-beamer-blank
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Info Beamer 1.0pre4.47a5ea+Lua 5.1 (http://info-beamer.org/)
Copyright (c) 2015 Florian Wesch <fw@dividuum.de>

[main.c] chdir /home/fenhl
[main.c] root node is info-beamer-blank
[main.c] tcp/udp port is 4444
[main.c] initial size is 1024x768
[info-beamer-blank] kernel: booting node
[main.c] initialization completed
[framebuffer.c] new framebuffer (1024x768): 1
[framebuffer.c] new framebuffer (1024x768): 2

Program received signal SIGSEGV, Segmentation fault.
0x00007fffea720472 in ?? () from /usr/lib/x86_64-linux-gnu/dri/nouveau_dri.so
(gdb) bt
#0  0x00007fffea720472 in ?? () from /usr/lib/x86_64-linux-gnu/dri/nouveau_dri.so
#1  0x00007fffea584594 in ?? () from /usr/lib/x86_64-linux-gnu/dri/nouveau_dri.so
#2  0x00007fffea44f48b in ?? () from /usr/lib/x86_64-linux-gnu/dri/nouveau_dri.so
#3  0x00007fffea3941e8 in ?? () from /usr/lib/x86_64-linux-gnu/dri/nouveau_dri.so
#4  0x000000000040903f in node_render_to_image (L=0x730fd0, node=0x627c40 <root>) at main.c:758
#5  0x0000000000406085 in luaRenderSelf (L=0x730fd0) at main.c:366
#6  0x00007ffff7bbb1f0 in ?? () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#7  0x00007ffff7bc5d42 in ?? () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#8  0x00007ffff7bbb64d in ?? () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#9  0x00007ffff7bba92e in ?? () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#10 0x00007ffff7bbb7bb in ?? () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#11 0x00007ffff7bb742c in lua_pcall () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#12 0x00000000004059d1 in lua_timed_pcall (node=0x627c40 <root>, in=3, out=0, error_handler_pos=7) at main.c:223
#13 0x0000000000405bd4 in lua_node_enter (node=0x627c40 <root>, args=3, bin=PROFILE_EVENT) at main.c:267
#14 0x0000000000405fbd in node_render_self (node=0x627c40 <root>, width=1024, height=768) at main.c:338
#15 0x000000000040fd4b in tick () at main.c:1450
#16 0x00000000004102b4 in main (argc=2, argv=0x7fffffffe128) at main.c:1585
```

dividuum, 2017-04-11:
Thanks. So it segfault [here](https://github.com/dividuum/info-beamer/blob/master/main.c#L758)? Or do you have a different version running?

fenhl, 2017-04-11:
I compiled the current master.

dividuum, 2017-04-12:
Hm. Odd. Do you have the closed source Nvidia driver available as well? Maybe I'm calling the OpenGL API wrong without noticing that causes the nouveau driver to segfault.

fenhl, 2017-04-12:
I can confirm that it no longer crashes after installing the closed source driver.