dividuum/info-beamer #60: Error using named arguments in load_video example

superlou, 2017-04-22:
When trying to play a video per the documentation tutorial, I receive `bad argument #1 to 'load_video' (string expected, got table)`.

I'm using the following code:

```
local video = resource.load_video{
  file="Macro_Plex_4K_Motion_Background_Loop.mp4"
}
```

I am using version 1.0pre4.47a5ea+Lua 5.1.  Is this expected behavior?

VelocityRa, 2017-04-22:
Haven't done Lua an a while but I think you're missing a semicolon in the second line.

dividuum, 2017-04-22:
Thanks VelocityRa, but the problem is not a syntax error in that case. The named arguments (through a table) isn't included in this version. But I think I can add support for that. Let me look at this.

dividuum, 2017-04-22:
May I suggest that you use the following instead?

```lua
gl.setup(1920, 1080)
local video = util.videoplayer("test.mp4")

function node.render()
    video:draw(0, 0, WIDTH, HEIGHT)
end 
```
Videos work quite a bit different when compared to the Raspberry Pi version: The Pi version decodes videos in the background, regardless of whether or not the video is displayed on screen. For the normal Linux version it works a bit different: The object returned through `load_video` returns a video object where you have to manually advance frames using the `:next` method. The wrapper code in `util.videoplayer` tries to abstract that away so you can load a video and then just `:draw` it somewhere and the video should play as expected. Would that work in your case?

superlou, 2017-04-22:
Does `util.videoplayer` work as a common API between the RPi and desktop Linux version, or are there other considerations?  I am hoping to do the bulk of my development on the desktop with testing on the Pi.

Using the above suggestion, I initially received an error about CPU usage being too high.  I transcoded the video to a 720P MOV file (dnxhd), and now it is somewhat playing.  The top half of the video is distorted, starting at the horizontal green band, and I receive `[video.c] incomplete video packet` in the terminal periodically.  I think this might be a matter of my transcoding, though.  Is there a recommended encoding that info-beamer prefers?

[image]

dividuum, 2017-04-22:
> Does util.videoplayer work as a common API between the RPi and desktop Linux version, or are there other considerations?

Kind of, but there might be subtle differences. The Pi version uses full hardware acceleration in a background thread while the Linux version software decodes a new frame each time a new `:draw` is requested. I'd highly suggest that you develop on the Pi version if that's your intended target. That way you can avoid surprises. The Pi version only supports the H264 codec as that is the only (reasonable) codec that is hardware accelerated on the Pi.

superlou, 2017-04-22:
That's fair.  I'm seeing some differences (no video looping) on the Pi, so I'll start getting familiar with developing on there.  Getting a little off-topic: while it's super bad practice to have a node load videos from a network location, is there a recommended way to develop on a Pi that's only accessible over the network?  SSH file system mounting seems to play havoc with the ability to automatically reload.

dividuum, 2017-04-23:
I guess mounting the Pi file system in some way on your development machine should work fine. That way the usual inotify handling for node/asset reloads should work. I guess you're using that sshfs that uses fuse? Which direction do you use that? Unless I'm overlooking something it should work when you mount the Pi from your dev machine.

superlou, 2017-04-26:
That does work.  For whatever reason, the editor I'm using (Atom) must be doing something a little funny, since if I use Xed or Vim, asset reloads work fine.

closed 2017-04-26 by superlou