dividuum/info-beamer #61: node.make_nested() missing in latest builds?

superlou, 2017-04-26:
When trying to run [the nested nodes example](https://github.com/info-beamer/nested-nodes-example), I get the following errors:

```
Info Beamer 1.0pre4.47a5ea+Lua 5.1 (http://info-beamer.org/)
Copyright (c) 2015 Florian Wesch <fw@dividuum.de>

[main.c] chdir /home/lsimons/workspace
[main.c] root node is nested-nodes-example
[main.c] tcp/udp port is 4444
[main.c] initial size is 1024x768
[nested-nodes-example] update +loader.lua
[nested-nodes-example] adding new child node example
[nested-nodes-example/example] update +text.txt
[nested-nodes-example/example] update +module.lua
[nested-nodes-example/example] update +silkscreen.ttf
[nested-nodes-example/example] kernel: booting node
[nested-nodes-example/example] runtime error: kernel.lua:411: cannot open file 'nested-nodes-example/example/node.lua'
stack traceback:
	[C]: in function 'load_file'
	kernel.lua:411: in function 'reload'
	kernel.lua:437: in function <kernel.lua:434>
[nested-nodes-example] update +README.txt
[nested-nodes-example] kernel: booting node
[nested-nodes-example] runtime error: nested-nodes-example/loader.lua:1: nested nodes feature missing
stack traceback:
	[C]: in function 'assert'
	nested-nodes-example/loader.lua:1: in main chunk
	(tail call): ?
	userlib.lua:531: in function 'require'
	nested-nodes-example/node.lua:3: in main chunk
	kernel.lua:393: in function 'load_into_sandbox'
	kernel.lua:410: in function 'reload'
	kernel.lua:437: in function <kernel.lua:434>
[main.c] initialization completed
[framebuffer.c] new framebuffer (1024x768): 1
[framebuffer.c] new framebuffer (1024x768): 2
```
Is this functionality deprecated?

dividuum, 2017-04-27:
On the contrary: It has only been added to the Pi version at the moment. The open source version is unfortunately falling behind a bit with regards to features. I don't have the time to backport them at the moment. Sorry.

superlou, 2017-04-27:
Is it something relatively straightforward that someone with some C background could implement as pull-request if given access to the RPi source?  Is the RPi source completely closed, or simply requires a license to use in commercial/public settings?

dividuum, 2017-04-27:
> Is it something relatively straightforward that someone with some C background could implement as pull-request [..]

Don't think so. While it seems like a single feature, it was a pretty big change that also required quite a bit of rewriting on the Lua side of things. The diff alone is 1500 lines.

> Is the RPi source completely closed[...]?

Right now it is. Companies might license it on request, but it's not free software. I'd love to change that somehow, but right now I don't see any other realistic way of financing continued development. I had flattr once and made 1.50€ :-}

superlou, 2017-04-27:
That's fair.  It's kind of a niche (but particularly helpful) product.  I'm hoping to sell my organization on a few licenses this year, and if it goes well, more the following.  I doubt that covers the man-hours to backport this.  Maybe I can convince the other folks looking into signage tech that it's a good excuse to pick up an RPi.

xstillwill, 2017-04-27:
It is :) we use it in Production on RPi 24/7.
Bringing passengers in our subway stations the joy of news and advertisements :D