dividuum/info-beamer #62: CPU usage too high

derchrisuk, 2017-05-03:
Hi,

I'm currently trying to get the Open Source version to run on an Atom 330 Quad-core CPU with i915 Graphics running on Ubuntu Server 17.04.
Applied some patches as in #57 and #41 prior to the compile.
After some reading I figured out I need X running as well.
So to keep it minimal I have no Window Manager running, just plain Xorg, and I run info-beamer like this as user 'flip':

```/usr/bin/xinit /usr/bin/dbus-launch --exit-with-session /usr/local/bin/info-beamer /home/flip/infobeamer_nodes/ -- :0 -nolisten tcp vt7```

This is the output I get:

```
X.Org X Server 1.19.3
Release Date: 2017-03-15
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.0-70-generic x86_64 Ubuntu
Current Operating System: Linux flipflop 4.10.0-20-generic #22-Ubuntu SMP Thu Apr 20 09:22:42 UTC 2017 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.10.0-20-generic root=/dev/mapper/flipflop--vg-root ro
Build Date: 28 March 2017  06:16:52AM
xorg-server 2:1.19.3-1ubuntu1 (For technical support please see http://www.ubuntu.com/support)
Current version of pixman: 0.34.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Wed May  3 16:10:28 2017
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
Info Beamer 1.0pre4.47a5ea+Lua 5.1 (http://info-beamer.org/)
Copyright (c) 2015 Florian Wesch <fw@dividuum.de>

[main.c] chdir /home/flip
[main.c] root node is infobeamer_nodes
[main.c] tcp/udp port is 4444
[main.c] initial size is 1024x768
libGL error: failed to open drm device: Permission denied
libGL error: failed to load driver: i915
[infobeamer_nodes] update +silkscreen.ttf
[infobeamer_nodes] kernel: booting node
[main.c] initialization completed
[framebuffer.c] new framebuffer (1024x768): 1
[infobeamer_nodes] timeout
[infobeamer_nodes] blacklisted for 60 seconds
[infobeamer_nodes] runtime error: kernel.lua:482: CPU usage too high
stack traceback:
        [C]: in function 'error'
        kernel.lua:482: in function <kernel.lua:481>
        [C]: in function 'render_self'
        kernel.lua:474: in function <kernel.lua:434>
[infobeamer_nodes] node is blacklisted
[framebuffer.c] new framebuffer (1024x768): 2
[infobeamer_nodes] node is blacklisted
[infobeamer_nodes] node is blacklisted
```

I'm using the simple Hello World example.
The 'node is blacklisted' is looping forever.
Not sure about the two libGL errors, and if they are causing all this.

derchrisuk, 2017-05-03:
I have installed a Gnome GUI in the meantime.
Enabled Debug, as a normal run would segfault.
This is the BT from gdb

```
(gdb) run /home/flip/infobeamer_nodes/
Starting program: /usr/local/bin/info-beamer /home/flip/infobeamer_nodes/
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Info Beamer 1.0pre4.47a5ea+Lua 5.1 (http://info-beamer.org/)
Copyright (c) 2015 Florian Wesch <fw@dividuum.de>

[main.c] chdir /home/flip
[main.c] root node is infobeamer_nodes
[main.c] tcp/udp port is 4444
[main.c] initial size is 1024x768
[infobeamer_nodes] update +silkscreen.ttf
[infobeamer_nodes] kernel: booting node
[main.c] initialization completed
[framebuffer.c] new framebuffer (1024x768): 1

Program received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x000055555555d15e in node_render_to_image (L=0x55555599c470, node=0x55555577d1a0 <root>) at main.c:719
#2  0x000055555555a0be in luaRenderSelf (L=0x55555599c470) at main.c:366
#3  0x00007ffff7bb53b0 in ?? () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#4  0x00007ffff7bc0062 in ?? () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#5  0x00007ffff7bb582d in ?? () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#6  0x00007ffff7bb4ade in ?? () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#7  0x00007ffff7bb599b in ?? () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#8  0x00007ffff7bb13f8 in lua_pcall () from /usr/lib/x86_64-linux-gnu/liblua5.1.so.0
#9  0x00005555555599a3 in lua_timed_pcall (node=0x55555577d1a0 <root>, in=3, out=0, error_handler_pos=7) at main.c:223
#10 0x0000555555559bd5 in lua_node_enter (node=0x55555577d1a0 <root>, args=3, bin=PROFILE_EVENT) at main.c:267
#11 0x0000555555559ff1 in node_render_self (node=0x55555577d1a0 <root>, width=1024, height=768) at main.c:338
#12 0x000055555556490c in tick () at main.c:1450
#13 0x0000555555564f43 in main (argc=2, argv=0x7fffffffe118) at main.c:1585
```

dividuum, 2017-05-04:
Might be the same problem as #59? What's in `main.c:719`?

derchrisuk, 2017-05-10:
Sorry for the delay.
Line 719 in main.c:

`glUseProgram(0);`