dividuum/info-beamer #65: Volume control

zontarian, 2017-06-21:
Hi,
our customer has bought 3 licenses of InfoBeamer, and all was well until they decided to change the specs, and now they want to be able to control the output volume of the video. 
We are using a RaspberryPi Zero as our video server.
I was wondering if, using OSC, it is possible to control the output volume. Until 10 minutes ago I wasn't even aware there was such a thing as OSC, and looking at the specs, it seems non trivial to understand.

Thanks

dividuum, 2017-06-21:
There is no support for changing the audio volume at the moment. Due to a [firmware bug](https://github.com/raspberrypi/firmware/issues/547) in the Pi, I'm looking at changing audio output to ALSA instead of OMX. In that case you could use the normal system mixer without talking to info-beamer at all. But that feature isn't finished yet.

You don't have to use OSC to talk to info-beamer. You can also use raw UDP packets. Here's an example of how to send data to info-beamer so you can use it in your Lua code: https://github.com/dividuum/info-beamer-nodes/tree/master/synchronized.

zontarian, 2017-06-21:
Hi Florian,
thanks. At the moment we use UDP packets to start videos (we have used your intermission sample as a starting point), but if I read your message correctly, you are suggesting to 

1. send custmo messages to LUA via UDP packets, with a custom simple protocol (i.e. volume up/ donw/mute)
2. in LUA, try to manipulate audio output level accordingly

But, since there's a bug in RaspPi, at the moment the player you use is not able to control it, and we would have to wait for your porting of the audio playback to ALSA.
But then you write that we can use the native system mixer, which means using directly commands available to the ALSA interface I gather.

But really we don't want to do this, since we  have used info beamer because, among many features, enabled us to have a remote playback server without writing a single line of code: info-beamer is in itself a server that handles commands, and it's scriptable.. so we would still like to talk to info-beamer. 
Do you think it will still be possible to control ALSA sound output from inside a LUA node in info-beamer?

Thanks

dividuum, 2017-06-21:
> 1. and 2.

Yes. That's how it might be done once volume control is possible from within Lua.

The firmware problem doesn't prevent me from changing volume levels - it's just not implemented at the moment. The bug prevents me from pausing a single video while another video is running, if (and only if) audio is used. That's why I want to switch to ALSA. But it's a bit complicated, so don't expect that in the near future.

> Do you think it will still be possible to control ALSA sound output from inside a LUA node in info-beamer?

I can't make any promises at the moment, as I'm not too familiar with how ALSA works. But if that's easily possible, I guess volume control might be a new feature.

zontarian, 2017-06-21:
ok, thanks for you candid reply.

I have another question: do you know if it's possible to control the volume output level of OMX via an external script in raspberry? we could issue a system command (but I don't know if it is possible from your LUA interpreter, and there don't seem to be an ad-hoc command)..
Just wondering, in my ignorance :)
thanks and again, compliments for the sleek product you have released.

dividuum, 2017-06-21:
I don't think that's possible as outputting audio using OMX is pretty low level. I don't think there's anything to adjust volume from "outside". Depending on how often you want to change volume and what output device you have connected: You might use CEC to adjust the volume of the connected HDMI screen. Not sure if that's viable or if that results in a visible volume slider from the screen itself, but it might be another option.

zontarian, 2017-06-21:
I'm sorry to bother you, but this opens up to me a whole world of unknown (till now) features and possibilities.. 
It seems I might be able to use `cec-client` from raspberry.. will it work while another process is using the HDMI ? And moreover, i remember vaguely that your LUA interpreter does not (of course) enable all LUA libraries, so `os.execute` will probably not work, is it correct?

Thanks again.

dividuum, 2017-06-21:
> It seems I might be able to use cec-client from raspberry.. will it work while another process is using the HDMI? 

Yes. I'm using it on the info-beamer hosted OS to the connected screen on/off on request. I'm not entirely sure about volume control, but it might work.

>  And moreover, i remember vaguely that your LUA interpreter does not (of course) enable all LUA libraries, so os.execute will probably not work, is it correct?

That is correct. You can't start external programs from within the info-beamer Lua environment.

novag, 2018-09-21:
Is there anything new regarding ALSA support?

dividuum, 2018-09-21:
Nope :-\