dividuum/info-beamer #67: Image/video animation

lunanigra, 2017-11-11:
Hello, currently trying an image/video animation. Sounds not too difficult; but seems I'm missing some basics.

Idea is having an image (e.g. a fish) moving round like one dot on this site...
http://www.creativecoding.org/example/processing:objektorientiertes-animieren-02-wandern

The calculation of the movement itself is easy. For every frame being rendered the position is slightly changed. But now it is necessary to calculate the right angle and rotate the image. Assuming it is a fish - the fish should look the same direction it is swimming.

There are some samples using shaders for rotation. So, a shader might be the right approach... Not sure... Would it be possible giving me some hints how to realise this idea?

Thanks, JC

dividuum, 2017-11-11:
No need for shaders:

```
for idx in #fishes do
   local fish = fishes[idx]
   gl.pushMatrix()
     gl.translate(fish.x, fish.y)
     gl.rotate(fish.angle, 0, 0, 1)
     gl.translate(-fish.width/2, -fish.height/2)
     fish.image:draw(0, 0, fish.width, fish.height)
  gl.popMatrix()
end
```
(Untested, as I don't have any fishes)

lunanigra, 2017-11-11:
Thanks! One further question... How do you initialize the fish or fishes object? Or what kind of object/structure is it?

dividuum, 2017-11-11:
[Tables](https://www.lua.org/manual/5.1/manual.html#2.5.7). Like every non-basic data type in Lua. Have a look at this example code for something similar: https://github.com/dividuum/info-beamer-nodes/blob/master/yoshi/node.lua

lunanigra, 2017-11-12:
:-)

closed 2017-11-13 by lunanigra