dividuum/info-beamer #68: Videos with alpha channel

lunanigra, 2017-11-13:
Hello, is that a chance playing videos with alpha channel?

I fear not as Raspberry PI only supports H.264. And H.264 does not support alpha channel.
If there is still a way please let me know which codec to be used.

Thanks, JC

dividuum, 2017-11-13:
As you've noticed: H264 doesn't support that. So the decoded video is always opaque. That said: If you render videos with the default GL video player (`resource.load_video` without the `raw=true` options), you get each video frame in a texture.

You can then render this texture onto the screen while a shader is active. Inside this shader you can test for a certain color and replace it with a transparent color. Basically "color keying". Here is an example shader that makes all "mostly" white pixels transparent:

```
local white_transparent = resource.create_shader[[
    uniform sampler2D Texture;
    varying vec2 TexCoord;
    uniform vec4 Color;

    void main() {
        vec4 col = texture2D(Texture, TexCoord).rgba;
        if (col.r + col.g + col.b > 2.9) {
            gl_FragColor = vec4(1.0, 1.0, 1.0, 0.0);
        } else {
            gl_FragColor = col;
        }
    }
]]
```

lunanigra, 2017-11-15:
Perfect!

closed 2017-11-15 by lunanigra