dividuum/info-beamer #69: How is the video output handled?

Sezotove, 2017-11-14:
Working on a project where we're looking to display videos/images and be able to send a stream back to a WOWza box. Thought I could do this is ffmpeg and grab x11, but realized it's not using x11 so that will not work.

Would you know the ffmpeg input option to use here?

So far the info-beamer platform is working well and handling what we need at the moment.

Thanks for the help.

dividuum, 2017-11-14:
I assume you're talking about the Raspberry Pi version, since that one doesn't use x11, right? I don't think there's any way to capture and encode a live stream of the Pi's output that's fast enough for any resolution other than potato quality and low frame rate. The only project I know that at least tries is https://github.com/DougGore/telepi. I personally didn't test it, but with it you should be able to capture the complete output and generate a h264 stream. But I seriously double that you get any thing better than 10fps with that.

Sezotove, 2017-11-14:
That is correct, it is the Pi version. From https://www.ffmpeg.org/ffmpeg-devices.html#Input-Devices it shows there are several methods of input that ffmpeg can grab, x11 being one of them. Was not sure if there were any others listed that were related to how you had the video output being shown.

Sezotove, 2017-11-14:
As an example, I thought maybe it was using framebuffer, so with ffmpeg I attempted to grab the /dev/fb0 and stream it, which worked, but only displayed my terminal, not the output from info-beamer.

dividuum, 2017-11-14:
That is expected, as info-beamer doesn't draw into the framebuffer. Instead it uses (multiple) dispmanx layers. See [this blog post](https://info-beamer.com/blog/raspberry-pi-hardware-video-scaler) if you want to learn a bit more about this. So x11 or framebuffer grabbing won't help. You'll have to snapshot the dispmanx output. The tool I linked does that. I don't know if ffmpeg is capable of that too. Unless there's Pi specific code, I doubt it.

closed 2017-11-16 by Sezotove