dividuum/info-beamer #71: Text over animated Background, how?

mheiste, 2018-09-13:
Hi,
i was wondering how ist is possible to draw fixed Text over animated background (like https://info-beamer.com/raspberry-pi-digital-signage-animated-background-pattern-138.html). As long as gl.perspective() ist used to get a fancy view, I can't use gl.push/popMatrix() to achieve this.

Currently my node.lua looks like this (taken from the link above):

Ideas?

`gl.setup(NATIVE_WIDTH, NATIVE_HEIGHT)

sys.set_flag("slow_gc")

util.resource_loader{
    "shader31c3.frag";
    "noise.png";
    "tiles.png";
}

local font = resource.load_font("RobotoSlab-Bold.ttf")

function node.render()
    local now = sys.now()

    local x = math.sin(now/62) * 320
    local y = math.cos(now/128) * 200

    gl.perspective(80 + math.cos(now/5) * 3, x + math.sin(now/14)*10, y + math.cos(now/14.4)*20, -100, x, y, 0)
    gl.rotate(math.cos(now/9)*5, 0, 0, 1)

    shader31c3:use{
        time = now / 1234.5;
        noise = noise;
        tile_res = {16, 16};
        res = {16.0 * 12, 16.0 * 12};
    }
    local size = 40
    tiles:draw(-16*size, -16*size, 16*size, 16*size)
    font:write(370, 340, "Hello", 400, 0.5,0.27,0,1)
end
`

Greets Mark ...

dividuum, 2018-09-13:
Try the following at the bottom of your `node.render` function:

```lua
   shader31c3:deactivate() -- deactivate shader
   gl.ortho() -- reset the projection matrix to its default
   font:write(370, 340, "Hello", 400, 0.5,0.27,0,1)
```

mheiste, 2018-09-13:
Works!

Thanks a lot!

closed 2018-09-13 by dividuum