dividuum/info-beamer #73: Support multiple fonts

fenhl, 2019-03-21:
I would like to request a feature where multiple fonts can be combined, with one acting as the main font and the others as fallbacks, similar to CSS.

The reason I'm requesting this is that [a project I'm working on](https://github.com/fenhl/info-beamer-twitch-chat) has a goal of displaying any text that can appear in a chat message, i.e. arbitrary Unicode. As a step in that direction, I want to switch to the [Noto](https://www.google.com/get/noto/) typeface, where each font is split into multiple TTF files since there is a hard limit of how many glyphs can be in a single TTF file.

Alternatively, a lower-level feature, such as a method on the font type to check whether it contains a glyph for a given character, could allow me to implement this myself.

Any solution would ideally to be available on both the Raspberry Pi version and the open-source version, since most of my info-beamer projects target both platforms.

dividuum, 2019-03-21:
How do multiple font files and switching between them work for other systems? Multiple files are open and if one doesn't provide a glyph it falls back to the next one?

fenhl, 2019-03-21:
That's how it works in CSS as far as I can tell.